UbiquityRobotics/ubiquity_main #6: Create and document NTP configuration

rohbotics, 2016-11-23:
We need a NTP configuration that we are on the road.

The RPi needs to sync off of a laptop timeserver without having to make that laptop have an internet connection (stratum problems).

@jim-v might have something

jim-v, 2016-11-23:
Install ntp on both machines:

`$ sudo apt install ntp`

On the server machine, insert the following into `/etc/ntp.conf`:

    server  127.127.1.0     # local clock
    fudge   127.127.1.0 stratum 10

The other machine can either use:

`$ sudo ntpdate serverhost`

or include the following into `/etc/ntp.conf`:

    server serverhost

rohbotics, 2016-11-23:
Great!
Can you make a PR that adds this documentation into the repo, I want to avoid using issues as docs if possible.

closed 2016-11-23 by rohbotics