UbiquityRobotics/ubiquity_main #8: Official Raspi Image

rohbotics, 2017-01-07:
Ubuntu 16.04
ROS Kinetic

rohbotics, 2017-01-30:
I am trying some image builds now, based on the ubuntu-pi-flavor-maker scripts.

It is a Lubuntu image with a bare bones version of ROS installed.

rohbotics, 2017-02-24:
ubiquityrobotics.com/2017-02-23-ubiquity-xenial-lxde-raspberry-pi.img.xz

rohbotics, 2017-03-04:
https://downloads.ubiquityrobotics.com/2017-03-01-ubiquity-xenial-lxde-raspberry-pi.img.xz

Is the latest and most stable, let me know if anyone has any issues with this image.

waynegramlich, 2017-03-05:
Rohan:

Please do not send a pointer to the .xz file.  The download page is what
should be sent out.  That way, when you create a new image there is a chance
that people will find it.  Many people will choose not to download the image
after reading the down page.

Thanks in advance,

-Wayne

rohbotics, 2017-03-05:
Good point, and that is the plan for all documentation links.
However, I am trying to specify this March 1st image as the one that could close this issue.

But yes, for pointing people towards a download page, please specify https://downloads.ubiquityrobotics.com.

anfederman, 2017-05-07:
Please add Intro-to-ros (SV-ROS)  and RBX1 (pirobot) to the catkin-ws/src directories,  These will aid by adding XV-11 and turtlebot tutorials to standard image.

anfederman, 2017-08-19:
Testing has shown several omisions in image file including fiducial follow and speech command parameter files.

rohbotics, 2017-08-19:
I closed this because the base image works, it boots and runs our stuff on boot up. 

Any other bugs should be opened as separate issues, with specific and detailed bug reports.

closed 2017-08-19 by rohbotics