UbiquityRobotics/ubiquity_main #11: Image Docuementation

waynegramlich, 2017-03-05:
This image contains:

- Ubuntu 16.04.02 LTS (Xenial Xarus) configured to run on a Raspberry Pi 3.
- Version ?? of the LXDE desktop.  Use of the LXDE desktop is optional and requires USB Mouse, USB Keyboard, HDMI display.
- The Kinetic Kame version of the ROS (Robot Operating System).
- Some packages specific to the Ubiquity Robotics Magni and Loki platforms.
- The Raspicam Node for interfacing either a version 1 or 2 of the Raspberry Pi camera module.

The image is encoded in Linux .xz format which is suitable for Linux systems only.  The commands for transferring this image are done from Linux shell running under Ubuntu 16.04 LTS as described below:

- Ensure that the needed tools are available:
  ```
  sudo apt-get install gnome-disk-utility
  sudo apt-get install wget
  ```

- Create a directory to work from:
   ```
   cd somewhere
   mkdir ubiquity_robotics
   cd rubiquity_robotics
   ```

- Download the image:
  ```
  wget get ...
  ```

- Run the `gnome-disks` utility:
   ```
   gnome-disks
   ```

   Rohan you fill in the rest.  I don't know how to use gnome-disks.

...

- Insert the micro-SD card into the Raspberry Pi 3.

- Power up the Raspberry Pi 3.

- Connect to the Raspberry Pi 3. via Wifi.

....

jim-v, 2017-05-07:
Please can you add:

sudo apt-get install screen ros-kinetic-tf2-web-republisher ros-kinetic-rosbridge-server ros-kinetic-nav-core ros-kinetic-move-base-msgs ros-kinetic-sick-tim

waynegramlich, 2018-01-20:
Superceded by the stuff in learn

closed 2018-01-20 by waynegramlich