UbiquityRobotics/ubiquity_main #23: setup-robot.sh fails when trying to find nav_core

davecrawley, 2017-05-17:
CMake fails to find nav_core on SD card comprising the downloaded image.

Full error message is:

CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "nav_core" with any
  of the following names:

    nav_coreConfig.cmake
    nav_core-config.cmake

  Add the installation prefix of "nav_core" to CMAKE_PREFIX_PATH or set
  "nav_core_DIR" to a directory containing one of the above files.  If
  "nav_core" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  move_basic/CMakeLists.txt:5 (find_package)

-- Could not find the required component 'nav_core'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "nav_core" with any
  of the following names:

    nav_coreConfig.cmake
    nav_core-config.cmake

  Add the installation prefix of "nav_core" to CMAKE_PREFIX_PATH or set
  "nav_core_DIR" to a directory containing one of the above files.  If
  "nav_core" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  move_basic/CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntu/catkin_ws/build/CMakeFiles/CMakeError.log".

jim-v, 2017-05-17:
I suggest running the command below manually until the image and/or script are updated:

```
sudo apt-get install screen ros-kinetic-tf2-web-republisher ros-kinetic-rosbridge-server ros-kinetic-nav-core ros-kinetic-move-base-msgs ros-kinetic-sick-tim
```

rohbotics, 2017-05-17:
Run that command, and then run `cd catkin_ws; catkin_make`

closed 2017-07-22 by rohbotics