UbiquityRobotics/ubiquity_main #30: URDF for loki needs to be modified

davecrawley, 2017-06-24:


rohbotics, 2017-12-23:
@waynegramlich can you put the measurements in this ticket? Thanks

waynegramlich, 2017-12-24:
# Notes for Rev. F

Remember, the Y axis is inverted in KiCAD.

## Sonar Locations

The locations of the sonars read from the KiCAD footprint locations are shown
in the table below.  The KiCAD angle is next to useless since the angle changes
from front to back.  The actual angle is more useful, where 0 degrees points along
the X axis.

<Table Border="1">
  <TR>
    <TH>Connector</TH>
    <TH>Side</TH>
    <TH>X(mm)</TH>
    <TH>Y(mm)</TH>
    <TH>KiCAD Angle(.1 deg)</TH>
    <TH>Actual Angle(deg)</TH>
    <TH>Comment</TH>
  </TR>

  <TR>
    <TD>N5</TD>
    <TD>Bottom</TD>
    <TD>349.5</TD>
    <TD>100</TD>
    <TD>-90.0(deg)</TD>
    <TD>0</TD>
    <TD>Faces Forward</TD>
  </TR>

  <TR>
    <TD>N6</TD>
    <TD>Top</TD>
    <TD>345.351435</TD>
    <TD>79.143753</TD>
    <TD>1125</TD>
    <TD>22.5</TD>
    <TD></TD>
  </TR>

  <TR>
    <TD>N7</TD>
    <TD>Bottom</TD>
    <TD>333.53732</TD>
    <TD>61.46268</TD>
    <TD>3150</TD>
    <TD>45.0</TD>
    <TD></TD>
  </TR>

  <TR>
    <TD>N8</TD>
    <TD>Top</TD>
    <TD>312.31</TD>
    <TD>55.215</TD>
    <TD>1575</TD>
    <TD>67.5</TD>
    <TD></TD>
  </TR>

  <TR>
    <TD>N9</TD>
    <TD>Bottom</TD>
    <TD>145</TD>
    <TD>45.5</TD>
    <TD>0(deg)</TD>
    <TD>90.0</TD>
    <TD>Faces Left</TD>
  </TR>

  <TR>
    <TD>N10</TD>
    <TD>Top</TD>
    <TD>124.143753</TD>
    <TD>49.648565</TD>
    <TD>2025</TD>
    <TD>112.5</TD>
    <TD></TD>
  </TR>

  <TR>
    <TD>N11</TD>
    <TD>Bottom</TD>
    <TD>106.46268</TD>
    <TD>61.46268</TD>
    <TD>450</TD>
    <TD>135</TD>
    <TD></TD>
  </TR>

  <TR>
    <TD>N12</TD>
    <TD>Top</TD>
    <TD>94.648565</TD>
    <TD>79.143753</TD>
    <TD>2475</TD>
    <TD>157.5</TD>
    <TD></TD>
  </TR>

  <TR>
    <TD>N13</TD>
    <TD>Bottom</TD>
    <TD>90.5</TD>
    <TD>100</TD>
    <TD>+90.0(deg)</TD>
    <TD>180</TD>
    <TD>Points backward</TD>
  </TR>

  <TR>
    <TD>N14</TD>
    <TD>Top</TD>
    <TD>94.648565</TD>
    <TD>120.856247</TD>
    <TD>2925</TD>
    <TD>202.5</TD>
    <TD></TD>
  </TR>

  <TR>
    <TD>N15</TD>
    <TD>Bottom</TD>
    <TD>106.46268</TD>
    <TD>138.53732</TD>
    <TD>1350</TD>
    <TD>225</TD>
    <TD></TD>
  </TR>

  <TR>
    <TD>N16</TD>
    <TD>Top</TD>
    <TD>124.143753</TD>
    <TD>150.351435</TD>
    <TD>3375</TD>
    <TD>247.5</TD>
    <TD></TD>
  </TR>

  <TR>
    <TD>N1</TD>
    <TD>Bottom</TD>
    <TD>145</TD>
    <TD>154.5</TD>
    <TD>180.0(deg)</TD>
    <TD>270</TD>
    <TD>Points Right</TD>
  </TR>

  <TR>
    <TD>N2</TD>
    <TD>Top</TD>
    <TD>312.61</TD>
    <TD>144.415</TD>
    <TD>225</TD>
    <TD>292.5</TD>
    <TD></TD>
  </TR>

  <TR>
    <TD>N3</TD>
    <TD>Bottom</TD>
    <TD>333.53732</TD>
    <TD>138.53732</TD>
    <TD>2250</TD>
    <TD>315</TD>
    <TD></TD>
  </TR>

  <TR>
    <TD>N4</TD>
    <TD>Top</TD>
    <TD>345.351435</TD>
    <TD>120.856247</TD>
    <TD>675</TD>
    <TD>337.5</TD>
    <TD></TD>
  </TR>
</Table>

## Motor Locations

The center point is between the two motors:

<Table Border="1">
  <TR>
    <TH>Connector</TH>
    <TH>X</TH>
    <TH>Y</TH>
    <TH>Comment</TH>
  </TR>

  <TR>
    <TD>N32</TD>
    <TD>271</TD>
    <TD>42.5</TD>
    <TD>Left Motor</TD>
  </TR>

  <TR>
    <TD>N33</TD>
    <TD>271</TD>
    <TD>157.5</TD>
    <TD>Right Motor</TD>
  </TR>

</Table>

## Camera Hole Locations

The four camera holes

<Table Border=1>
  <TR>
    <TH>Hole</TH>
    <TH>X</TH>
    <TH>Y</TH>
  </TR>

  <TR>
    <TD>H5</TD>
    <TD>126</TD>
    <TD>103.5</TD>
  </TR>

  <TR>
    <TD>H6</TD>
    <TD>113.5</TD>
    <TD>103.5</TD>
  </TR>

  <TR>
    <TD>H7</TD>
    <TD>126</TD>
    <TD>82.5</TD>
  </TR>

  <TR>
    <TD>H8</TD>
    <TD>113.5</TD>
    <TD>82.5</TD>
  </TR>

</Table>

rohbotics, 2017-12-26:
Thanks Wayne, I made a PR to ubiquity launches to update the URDF.

waynegramlich, 2017-12-26:
In many practitioners of software engineering that believe in the philosophy that if something is hard to write it should be equally hard to read. Alas, I do not subscribe to that philosophy.  The `rpy` field for each of the sonar rotations should be of the form `rpy="0 0 R"`.  I feel that all entries of the form `rpy="3.1415 0 R"` are confusing.  A comment that shows the rotation in degrees would be helpful as well. `` for the `rpy="0 0 3.1415926"` entry.  The mesh entries are all commented out.  Perhaps doing a simple mesh where the sonar is represented as a block?

jim-v, 2017-12-26:
It should be possible to get rid of the 180 degrees rotation around the `x` axis by negating the `y` component of translation.

rohbotics, 2017-12-26:
The `rpy="3.1415 0 R` is for the sonars that are upside down on the bottom of the loki. I just clarified this in a comment.

@jim-v In URDF the rotation is about the center of the link (not the joint origin) so is independent of translation.

Fixed the mesh issue.

waynegramlich, 2017-12-26:
I have looked for the comment and could not find it.  The comment should specify the origin and orientation of the sonar `.stl` file.  The comment should mention that a rotation of 180 degrees along the X axis is used to flip the sonar to the bottom surface of the PCB.  I still encourage that the Z axis rotation be supplied as a comment in degrees.

rohbotics, 2017-12-26:
@waynegramlich I missed a commit/push. I added the Z axis rotation as well.

waynegramlich, 2017-12-27:
I'm happy, if you are happy.  I'd close the issue out, but I'd probably screw up.  So, I'll let you close the issue.

closed 2017-12-27 by rohbotics