UbiquityRobotics/ubiquity_main #43: Move-base on loki to enable speech recognition

davecrawley, 2017-06-29:


jim-v, 2018-01-15:
The odometry looks good, but the rotation does not slow down as the target orientation is reached. This appears to be related to UbiquityRobotics/loki_base_node#2 in that there appears to be an issue with motor speed requests being handled by the firmware.

rohbotics, 2018-01-15:
Could it also be https://github.com/UbiquityRobotics/loki_base_node/issues/1? 

With teleop it can fall behind and lag on executing commands, maybe that is causing the issue.

jim-v, 2018-01-15:
That's part of the problem, and it does continue to rotate for a few seconds after the last command was sent.   However, changing the rate of the rotate loop from 50 to 5 did not cause it to behave correctly.
It can be made to behave better by setting the `min_angular_velocity` parameter to 0.5

jim-v, 2018-01-15:
The CPU usage of the loki_base_node process is around 12% when idle and 40% when going to a goal.

waynegramlich, 2018-01-15:
Assigning this task to me is probably not optimal. I no longer have a Loki.  It sounds like there is a fundamental bug in the software stack.  It could be loki_base_node or it could be the firmware.  The first step is to verify that the PID loops are working correctly.  This is done using `minicom` (or equivalent) to send `m left right` commands to the Loki.  The `e` command returns encoder values.  This response is pathetic, but that is the best I can suggest right now.

closed 2018-02-28 by jim-v