UbiquityRobotics/ubiquity_main #65: Design issue: Rosbridge ws or wss?

jrlandau, 2017-08-17:
When running Speech in the browser, Chrome requires https which in turn requires wss. This requires rosbridge to be set up to require a wss connection, with a cert.  Robot Commander in Android does not require all this.  The question is, shall we allow the two to operate differently (how?) or should we force Robot Commander to use wss with a rosbridge that requires it?

anfederman, 2017-08-19:
on default image, I see joystick launch, but not speech command launch, also by default I see fiducial follow is not being started, therefore the camera and aruco are not being started either. Is that by design or oversight?

anfederman, 2017-08-19:
Oversight -  magni_demos speech control launch refers to files /etc/letsencrypt not present in default image.

jrlandau, 2017-08-19:
let's discuss at the huddle today

rohbotics, 2017-08-19:
@anfederman I think you meant your first comment for #64. This issue is about ws vs wss and how to support either.

Also @anfederman that is why the SSL is support is commented out in the launch file, because the SSL certs aren't there. More on this later in this message.

We could run both a ws and wss version of rosbridge on different ports, and have the android app and the website use different ports when connecting to the websockets.

But this still has the problem of where do the certs come from. A cert needs to be issued to a specific globally registered domain name. The problem with the robots is that there is no one domain name to connect to, as every robot will be on a different IP and may not even have an globally registered name.

For a brand new robot still in AP mode, we do have a globally registered domain, robots.ubiquityrobotics.com, and we could provide trusted SSL certs/keys for this, but when you switch to non AP mode, this domain will no longer work. We could ask users to change their /etc/hosts to make this work, but that is a really ugly solution. 

The other way to make this work is to use self-signed certs and make sure that users understand to click through the big red security warning (not ideal).

rohbotics, 2017-08-20:
Here is what we decided at the huddle today:

- Run 2 instances of rosbridge one using ws on port 9090 and one with wss on 9443.
- Use self signed certs, and document the security work around
- Have a letsencrypt setup for demos.

davecrawley, 2017-08-21:
Is there any way we can place the self signed certs key in the browser / phone white-list? That way there will be no security warning? That way we'd have to document the white-listing procedure, but not that they have to ignore the security warning every time they connect.

rohbotics, 2017-08-21:
@davecrawley 

We could put up instructions for how to trust the cert into a computer, but that can become really complicated for a phone (especially for Android 7.0+).

jrlandau, 2017-08-23:
@davecrawley <https://github.com/davecrawley>

So what is the resolution of this issue?

which raises an interesting question:  does stuff like this get documented?

davecrawley, 2017-08-23:
Do we need it for the phone? I thought the root cause of the architecture we chose was so that we could do this on a laptop. Can we document how to trust the cert on a laptop and do it another way for a phone?

jrlandau, 2017-08-23:
@davecrawley
I do not understand what you are asking here. If it is me whom you are
asking, please telephone me.

If not, please let me know what has been decided, without using any
pronouns. At present I do not know what to implement.

rohbotics, 2017-08-23:
Current state of decision: 

- Run 2 instances of rosbridge one using ws on port 9090 and one with wss on 9443.
- Make the app use port 9090 and the website use port 9443.
- Use self signed certs, and document how to set up trust on the cert on a laptop.
- Have a letsencrypt publicly trusted SSL setup for demos.

Definition of Done for this issue:

- [ ] Consensus on the design decision we are making here.
- [ ] Issues opened on relevant repositories to implement the decision.

jrlandau, 2017-08-23:
How is one to go about creating those 2 instances of rosbridge?  Is it done
via launch files? If so, where on github are the launch files?

closed 2017-09-09 by rohbotics