UbiquityRobotics/ubiquity_main #66: fiducial_follow not running after robot boots

jrlandau, 2017-08-22:


rohbotics, 2017-08-23:
We cannot have both fiducial follow and navigation running at the same time. We also decided that we son't want to run fiduical follow until the user understands what it does, and how it could make the robot move unexpectedly.

closed 2017-08-23 by rohbotics

jrlandau, 2017-08-23:
I think we should revisit this issue.  Maybe there is a way to avoid
burying it.

davecrawley, 2017-08-23:
What would the user expect? Or perhaps I should ask, what would delight the user most? Remember we have lots of options on fiducial follow. The last I recall was that we were going to make use of the two action buttons, one for provision and one for a demo mode. Also the last thing that I recall - that doesn't conflict with what Rohan said - but perhaps has a different emphasis, is that we were going to have a special fiducal for fiducal follow, that would be downloadable with a warning about the effect it can have on the robot. Anyways it doesn't particularly matter what I think, nor what anyone things, its what will delight the user the most.