UbiquityRobotics/ubiquity_main #81: voice commander rotate issue

davecrawley, 2017-11-21:
When you give it a specific rotate command (e.g. rotate right 10 degrees) it rotates continuously if it is not able to determine its pose through robot commander.

jim-v, 2017-11-21:
What is the output of `apt-cache policy ros-kinetic-move-basic`?

jim-v, 2017-11-21:
I am have not been able to reproduce this.  I ran `roslaunch magni_demos speech_control.launch`, connected via the RobotCommander app and gave the verbal command "rotate right 10 degrees".  The robot rotated about 10 degrees.

```$ apt-cache policy ros-kinetic-move-basic 
ros-kinetic-move-basic:
  Installed: 0.2.2-1xenial-20171008-204812-0800
  Candidate: 0.2.2-1xenial-20171008-204812-0800
  Version table:
 *** 0.2.2-1xenial-20171008-204812-0800 500
        500 http://packages.ros.org/ros/ubuntu xenial/main armhf Packages
        100 /var/lib/dpkg/status
```

jrlandau, 2017-11-21:
Just an fyi--measured movement commands like rotate right 10 degrees don't
actually determine the pose in RC--RC just sends a command to move_base to
move that much from wherever it is now.

jim-v, 2017-11-21:
Also, navigation to relative goals is performed using the robot's pose in the `odom` frame, which comes from the motor node. The robot's pose in the `map` frame is used to navigate to absolute goals.

davecrawley, 2017-11-21:
Is this the right way to do it?

Seems like we've covered the forward and back motion that way, but we'd
need to cover rotate too individually if we do it that way. No?

rohbotics, 2017-11-22:
@davecrawley I am not sure I understand your question.

When move_basic gets a "relative goal" under the frame base_link, it transforms it into the odom frame and  then attempts to navigate to that point, using the standard arctan then rotate to pose behavior. 

For a simple rotate command, this would mean to arctan motion (no linear displacement needed) so it will go right to the final rotate to pose.

davecrawley, 2017-11-22:
OK, we may be speaking at cross purposes, perhaps we should get back to the
original problem.

Whenever we set a specific rotate goal (e.g. rotate right 10 degrees) when
the robot didn't see fiducials it would rotate continuously without
stopping. There is a bug somewhere. I probably shouldn't hypothesize too
much as to what the problem is nor how to fix it. But would be very happy
to try to reproduce it in the company of someone more knowledgeable than me
and then we can jointly think about how to solve it.

David

davecrawley, 2017-11-22:
Jim,

When you tried to reproduce this, was the robot seeing fiducials or not?

David

jim-v, 2017-11-22:
@davecrawley No.
 
Can you please run `apt-cache policy ros-kinetic-move-basic`  If you are getting this issue with the 0.2.2 version, I will look into it.

jim-v, 2017-11-22:
It looks like a commit in September https://github.com/UbiquityRobotics/move_basic/commit/e15bbbf37b88d9efd85bfa123dda360920ff1b15#diff-25d902c24283ab8cfbac54dfa101ad31
was not merged, so `move_basic` is not working as intended with relative goals.  I will rebase and test it and submit a PR.

rohbotics, 2017-11-23:
@jim-v 

This seems to be a firmware bug, the odometery on the robot that David was testing on was way off, the robot moved a few meters and the translational x only changed by about a millimeter. 

This seems to be due to incorrect sending of odom data down the wire, probably because of recent firmware changes for the new hardware.

davecrawley, 2017-11-24:
In the last revision of the firmware we attempted to fix the screwy wiring
order required with sensing wires from the left side of the PCB feeding a
motor the power of which was wired to the right side of the PCB. We rewired
so that the wheels would turn with the sense wires correctly connected. The
result was that the motor thought the wheels were on the wrong way round
with one wheel going backwards - a problem we fixed so that the robot would
go straight when you told it to go straight.

However we didn't make a parallel change in the odom so it thought that the
left was the right and that one wheel was going in the wrong direction.

I have now fixed this in firmware - but I am not sure I've done it right.
I'd really like to :

1) Have Jim review my code
2) Confirm that I have done it right with someone.

I kind of feel like we need to do that in person with the actual robot
being used.

Jim I appreciate you are tied up right now, so it may not happen right
away. When would be a good time to get together to do this? I can come to
you?

David

jim-v, 2017-11-25:
@davecrawley, I'll see comments in an issue more quickly if you mention @jim-v.

To test if the odometry is making it to ROS:

`rosrun tf tf_echo odom base_link`

You should see something like this:

```
 Translation: [0.000, 0.000, 0.000]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
            in RPY (radian) [0.000, -0.000, 0.000]
            in RPY (degree) [0.000, -0.000, 0.000]
```

Rotate in place with teleop, and you should see rotation around the _z_ axis change:

```
- Translation: [0.008, 0.002, 0.000]
- Rotation: in Quaternion [0.000, 0.000, 1.000, 0.026]
            in RPY (radian) [0.000, -0.000, 3.090]
            in RPY (degree) [0.000, -0.000, 177.063]
```

Driving forward and backward (with zero rotation) should cause the _x_ component of translation to change:

```
- Translation: [0.867, -0.048, 0.000]
```

davecrawley, 2017-11-26:
OK @jim-v

So I ran that test. Driving forward with teleop does indeed cause the x
component to change. I can drive it forward about 1 meter and the rotation
component only changes by a degree or two. Couple of questions.

1) The x component becomes more negative when driving the robot forward. Is
this correct if it isn't the firmware fix is easy.

2) If this is correct when I attempt to drive the robot using voice
commander by setting waypoints it goes to completely the wrong place. It
doesn't look like its driving backwards it looks like it is confidently
going to and utterly incorrect place. Any ideas?

David

rohbotics, 2017-11-26:
@davecrawley That is incorrect behavior on the odom topic, make sure that the linear x increases as the robot moves forward, and the rotational z increases rotating counter clockwise. 

Without that precondition, move_basic behavior will be pretty weird, as you are seeing.

davecrawley, 2017-11-26:
So I reversed the firmware direction, and everything seems to be working
OK!

The code requires some significant cleanup as all the variable names and
pin names are a bit illogical (e.g. M1 things tied to M2 etc.) - but the
code works.

David

closed 2017-12-23 by rohbotics