UbiquityRobotics/ubiquity_main #82: default .bashrc files for all ubiquity computers running ROS

anfederman, 2017-11-21:
It has been my bitter experience that using  computer.local in ROS_IP and ROS_MASTER_URI is subtlety disastrous, in that topics get listed but don't echo across network connections.

Therefore I always explicitly set these environment variables. I do this in my .bashrc files on both robot and remote.

Laptop:

source ~/catkin_ws/devel/setup.bash
ip="$(hostname -I|cut -d ' ' -f 1)"
export ROS_IP=$ip

on the remote I add one additional line to the .bashrc:

export ROS_MASTER_URI=http://$ip:11311

Failure to do this makes your life miserable.  For example yesterday at David's working with a fresh image, keybord teleop didn't work until the ROS environment variables were reset to IP number.

jim-v, 2017-11-21:
One should set `ROS_HOSTNAME` if using a name and `ROS_IP` if using an IP *address*.

http://wiki.ros.org/ROS/EnvironmentVariables#ROS_IP.2BAC8-ROS_HOSTNAME

jrlandau, 2017-11-21:
Is this something we want all our users to always do?  If so, is there a
way to do it for them?  Or must we tell them to do it in the documentation?

I think that the odds of getting people to type that lovely command line of
Alan's are pretty discouraging.

jim-v, 2017-11-30:
@jrlandau The idea is not to type it, but to put it in the user's `.bashrc` file.

It could be done on one line:

```
export ROS_IP=$(hostname -I | cut -f1 -d" ")
```
with a  corresponding:
```
export ROS_MASTER_URI=http://${ROS_IP}:11311
```
However, the advantage of using names of IP addresses is that the nodes on a machine can all talk to each other, even if the interface goes down.

jrlandau, 2017-11-30:
@jim-v , how would we get that line into the user's .bashrc file?

closed 2017-12-21 by rohbotics