UbiquityRobotics/ubiquity_main #84: Auto Detect Magni v Loki

rohbotics, 2018-01-06:
A script to determine whether we are a magni or loki. 

Probably will be implemented through serial port probing.

waynegramlich, 2018-01-07:
I need the exact sequence of bytes to be sent to the Magini and what is supposed to come back.  I also need the baud rate.

davecrawley, 2018-01-09:
Are we also going to detect if its not connected to either a Magni or Loki i.e. just a bare Raspberry Pi? 

There are plenty of legitimate uses of our image in that configuration too. 

David

rohbotics, 2018-01-09:
Yes this needs 3 states, Magni, Loki and Neither. With neither, the auto start needs to be disabled.

waynegramlich, 2018-01-09:
For sure, I will add the code to Loki to talk the binary protocol somehow.  I still need the exact byte
sequences for Loki and Magni.  Once the Loki firmware is updated, it should be easy to write a quick Python program that performs the query.

jim-v, 2018-01-09:
@waynegramlich see https://github.com/UbiquityRobotics/ubiquity_motor/blob/indigo-devel/src/motor_hardware.cc#L197 for how to send a firmware version request and https://github.com/UbiquityRobotics/ubiquity_motor/blob/indigo-devel/src/motor_hardware.cc#L101 for receiving the response. 
https://github.com/UbiquityRobotics/ubiquity_motor/blob/indigo-devel/src/motor_message.cc deals with encoding and decoding the messages.  Note that messages besides the response to your request may come in, so you will need to look at the type.  It runs at 38400 baud.

waynegramlich, 2018-01-09:
I hate to be a stick in the mud, but is the protocol documented anywhere?  Yeah, I can wade through all of the C++ code, but it is really easy to miss something.

jim-v, 2018-01-09:
I don't know about documentation, I thought there was some, but I can not
find it.

Another possibility is that we have the magni and loki packages provide an
executable that will check for the hardware in question, reusing the
existing code, then the script just needs to call them both.

davecrawley, 2018-01-10:
There is documentation covering the protocol - but the documentation is
deprecated as the protocol has evolved and the documentation hasn't kept up
- in particular this feature was not included in them, so you'll do a lot
better following the C. However Jim perhaps you could write a short
description of what a packet looks like. As I remember it something like:

8 byte packet

1 leading edge byte that contains character ????? (what is the leading edge
character? I know we picked it with some intelligence but I can't remember
what it was)
3 bytes of commands
4 bytes of data

Jim I think you might know this better than I, can you write a short
summary similar to the above that will help Wayne?

David

rohbotics, 2018-01-10:
The command to send over the link is the request for hardware version (register 21).

`0x7E 0x3A 0x21 0x00 0x00 0x00 0x00 0x26`

The response will take the form of:  
`0x7E 0x3B 0x21 4_DATA_BYTES CHECKSUM`

Checksum spec:
```
Step 1: Compute the sum of all characters without including the command delimiter and the checksum:
Sum = 02+0A+F3+C2+D3+3E+4F
         = 321
Step 2: Keep the last eight bits and discard the rest:
= 21
Step 3: Subtract the result of step 2 from FF:
	= FF - 21
	= DE
This result is the checksum.
Step 4: Insert the checksum to the end of the packet:
	Therefore, the final packet becomes: 7E 02 0A F3 C2 D3 3E 4F DE
```

I don't think that there is any spec for the format of the hardware version register.

I propose that the first byte is for the robot type (With Magni as 0x00 and Loki as 0x01), and the other 3 bytes for the revision, this way there is no firmware changes for Magni.

rohbotics, 2018-01-10:
psuedo code for calculating checksum:

```
sum = 0
for byte in command:
     sum += byte
sum = sum & FF
return FF - sum
```

waynegramlich, 2018-01-10:
Life would be a lot easier if the spec. were actually written down in the code some place.

RasPi=>Loki:
    0x7E 0x3A 0x21 0x00 0x00 0x00 0x00 0xA4

Loki=>Raspi:
    0x7E 0x3B 0x21 0x01 0x00 0x00 0x00 0xA2

This takes place using an 8N1 serial link running at 38400 bits per second.

Um.  Why are you running the link so slow?  Loki currently runs at 115200.  There
are 16 sonars blasting away on Loki and each comes back with an id, distance, and
timestamp.  That does chew up some bandwidth.

rohbotics, 2018-01-10:
Documenting the Protocol in the code is probably a good idea.

The byte sequences you have posted look good, but since we are loki rev F, it the data on Loki -> Pi should be `0x01 0x00 0x00 0x06`. 

The link is that slow because the PSOC didn't have a crystal.... so any faster would cause timing issues...

I don't know if this is still the case, @davecrawley ?

waynegramlich, 2018-01-11:
If we are serious about this, the first byte should be the platform (0x00=>Magni, 0x01=>Loki),
the second byte should PCB revision, and the last two bytes should be the firmware version.
I propose:
     0x01 (Loki) 0x06 (Rev. F) 0x00 0x00 (Firmware version)

rohbotics, 2018-01-11:
Register `0x22` is specified as Firmware Version, so the firmware stuff should go there. (We get a whole 32 bits for it). Right now Magni returns firmware version 24.

I shared the specification google doc with you. The only change since the spec was written is that the protocol version and Read / Write / Response and bytes have been combined. The version is in the upper 4 bits of the 2nd byte, and the Read / Write / Response is the lower 4 bits. Also protocol version has been bumped to 3. 

The register map in the spec is correct.

davecrawley, 2018-01-11:
The PSoC can run without a crystal but the internal oscillator is lousy
and, as Rohan points out affects external communication. We put in a
Crystal several board revs ago when this problem was identified, however we
never updated the software. Because the data requirements aren't as
demanding as on Loki its one of those, good to do, not essential to do
things. I am sure we will update protocol speed when we start looking at
firmware again.

Apart from anything, doing an OTA update at 32kbaud is going to suck.

David

waynegramlich, 2018-01-11:
Baud Rate Issue: Platform discovery will be much more reliable if both Loki and Magni use the same
baud rate.  The ATMega2560 running with a 16MHz crystal kind of maxes out at 115200.  Since that is where most Linux serial drivers kind of whimp out, I would like to recommend upgrading Magni to 115200 rather than downgrading Loki to 38400.

Register numbers: As Rohan points out , the specification (which he sent to me thoughtfully sent to me) provides resgister 0x21 for the hardware version and 0x22 for the firmware version.  Works for me.  The clean thing to do is add another register for platform id.  Lacking that we need to encode the platform id and PCB id 0x21.  I prefer doing it right and adding another register.  0x43 is available.  I would specify 0 for Magni and 1 for Loki.

Eventually, I can add all of the appropriate other registers to the Loki firmware and we can deprecate the serial Loki_Node and use a common Loki/Magni node.

rohbotics, 2018-01-11:
I agree we should use 115200 8N1 on both Magni and Loki. 

@davecrawley @jim-v 
How much effort would it be to up the baud rate and add a robot id register to the PSOC?

@waynegramlich I misspoke (mistyped?) there have been some additional registers added since the spec was updated. The next available register is 0x31.

        REG_BOTH_SPEED_SET = 0x2A,
        REG_MOVING_BUF_SIZE = 0x2B,

        REG_LIMIT_REACHED = 0x2C,
        REG_BOTH_ERROR = 0x2D,
        REG_BOTH_ODOM = 0x30,

        DEBUG_50 = 0x50,
        DEBUG_51 = 0x51,
        DEBUG_52 = 0x52,
        DEBUG_53 = 0x53,
        DEBUG_54 = 0x54,
        DEBUG_55 = 0x55,
        DEBUG_56 = 0x56,
        DEBUG_57 = 0x57,
        DEBUG_58 = 0x58

I will put some effort into updating the spec tomorrow night. 

Before we can use the same protocol on both, we need to add a timestamp to the messages that go MCU => Pi, we want to keep Pi => MCU only 8 bytes (so the whole message can be read using just the hardware buffers in the PSoC, but we can spare bytes for MCU => Pi. 

Additionally we will need to add sonar support to the protocol.

waynegramlich, 2018-01-11:
Updating the protocol spec. is a good idea.  One of these years, Google is going to loose interest in Google Docs, so I recommend storing it in the Magni Node repo.

If you are going to do protocol work, you should actually fix the protocol so there is some way to transmit a data bytes that contains 0x7e 02.  In general, designing a protocol around a hardware limitation of the PSOC is not a particularly good practice, since eventually the PSOC gets replaced, but the protocol does not get replaced.  If you have ever seen the FRAM oil filter commercial, they point out "pay now or pay later", where the "pay later" option is 3 orders of magnitude more expensive.

davecrawley, 2018-01-11:
Agree having baud rate be 115200 on both platforms is the right idea. How
much effort? If no problems are encountered along the way its a single 1-2
hour session kind of effort. Last time we tried this we had the slightly
sloppy timing on the PSoC but also slightly sloppy timing on the RPi - the
two together is what made 115200 difficult. As I recall the slightly sloppy
timing on the RPi is systematic, in that the RPi 115200 isn't exactly
115200 its always off by a certain amount so theoretically could be cured
by "tuning" the speed on the PSoC to match the actual speed on the RPi.
We'd do a session in Rohan's place as his scope has a serial debugger.

8 byte protocol matters. This is not particular to PSoC, Atmel has the same
buffer size,  and is a pretty common hardware buffer size for uC. The
Practice should be, design your protocol so that it fits in the hardware
buffer of common uC. If you don't keep the protocol to the hardware buffer
then unless the uC has a MMU (which almost no low end uC do) then the uC
raises an interrupt and waits while all 9 (or 10 or 11 bytes) come in. 9
bytes => 72 bits   => 0.6ms @ 115200 baud or  2ms @ 38400 baud. In practice
this will limit the speed with which you can run your PID loop. On Magni we
run our PID loop at 10kHz so a 2ms or even a 0.6ms halt is pretty deadly.
With hardware serial the entire message is stored on the serial buffer and
then transferred over to processor memory on the chip's internal bus
(ususally that will be AXI in most uC these days) which is a wide high
speed bus that can transfer the entire thing over in like 2 clock cycles. I
am sure there are other ways to get around the hardware limitation (like
raising an interrupt for each byte) but seriously, just keep the protocol
nice and tight and in packets of less than 8 bytes. We should be doing that
anyway.

David

waynegramlich, 2018-01-11:
My first reaction to reading that the RiPi3 has flaky UART's was "I doubt it."  However, I was wrong.  Somebody wrote up an [informative article](https://spellfoundry.com/2016/05/29/configuring-gpio-serial-port-raspbian-jessie-including-pi-3).  I strongly recommend that we use the hardware UART instead of the software UART.

The next paragraph about the protocol exposes many flaws in the design and implementation.  I don't have time to go into all of the problems.  The short summary is below:

1. PID loop:  You are running your PID loop too fast.

2. UART interrupts: It is a really good idea use UART interrupts.

3. Protocol Efficiency: It is fairly easy to design a better protocol that still fits in 8-bytes.

We can talk about it more on Saturday.

davecrawley, 2018-01-11:
> My first reaction to reading that the RiPi3 has flaky UART's was "I doubt
> it." However, I was wrong. Somebody wrote up an informative article [...]
The next paragraph about the protocol exposes many flaws in the design and
> implementation. I don't have time to go into all of the problems. The short
> summary is below:
>
Look, I know we could have done single byte transmission with interrupts
for each byte. I know we could have done a time division architecture. We
didn't do those things with good reasons.

>
>    1.
>
>    PID loop: You are running your PID loop too fast.
>
> Well if we were running a speed oriented PID loop. I'd agree with you, but
we aren't. We arrived at the current position oriented method after lots of
experimentation to get better performance from the robot and it works
dramatically better than speed oriented methods. This method works and
works well as it avoids the usual problems of discretization errors that
mean that you can't run a speed oriented PID loop too quickly. Because our
position oriented control method doesn't suffer from these problems
performance improves the faster you run the loop with a theoretical upper
limit determined only by the size of the register for your integral term
and the speed of the uC. In this configuration if the PID loop critical
damping time is shorter than the speed with which the robot is capable of
generating ticks then the wheel will always be within 1 tick of its ideal
position provided the PID loop is critically damped (e.g. if the PID loop
is properly tuned). Anyway I can explain this and the rationale behind it
more on Saturday.

>
>    1.
>
>    Protocol Efficiency: It is fairly easy to design a better protocol
>    that still fits in 8-bytes
>
> I violently agree with this one.

David

waynegramlich, 2018-01-12:
I forgot you were doing position based.

The AVR USART has a 2 byte receive buffer.

We'll talk about it some more.  I may not be able to make this weeks Sat. huddle.

rohbotics, 2018-01-13:
Updated serial protocol doc:
https://github.com/UbiquityRobotics/ubiquity_motor/blob/indigo-devel/Serial_Protocol.md

rohbotics, 2018-01-13:
I think this discussion got de-railed a little bit.

I think it is reasonable to simply use the hardware version register, especially with all the old magnis still in use, which wouldn't get the new register.

@waynegramlich How much work would it be to get Loki to respond to a request for register 0x21, with the first byte as the robot id?

How much effort do you think it would be implement Magni/Loki/Neither detection with the mismatched baud rates? If it isn't too difficult to handle, I can make a program in ubiquity_motor that sends out the message, reusing all the protocol handling code.

waynegramlich, 2018-01-13:
If the baud rates do not match, there is no need for Loki to use the Magni protocol.  To test for Magni, send whatever message you want at 38400.  To test for Loki, open 115200 and send "b\r" and see if you get back "115200".  A timeout is required, since the baud rate mismatch will cause garbled stuff to show show up on one platform or the other.  If the error checking on either platform is weak, bad things will happen.  I think this is a really bad solution.

rohbotics, 2018-01-13:
Ok, so we should do the baud rate bump to 115200 on Magni.
Since this breaks compat, we can easily add a new register for robot id, `0x31`.

For future protocol unification, we should get together and design a protocol we can all live with before shipping, but I would but the auto detection stuff at a higher priority than that right now. 

Right now we are getting (~10-20) image downloads per day, so I want to get the no robot mode working by the February image.

rohbotics, 2018-01-13:
From the huddle:

- [ ] Baud Rate and Robot Id Register on Magni (@jim-v @davecrawley)
- [ ] Robot Id register on Loki (@waynegramlich)
- [ ] RPi side code to acutely read the register (@rohbotics)

waynegramlich, 2018-01-14:
It looks like the register is going to be 0x31.  Returns 0x00 for Magni and 0x01 for Loki.  @rohbotics should update the protocol specification document.

waynegramlich, 2018-01-15:
OK. I generated a pull request that has the firmware updates to support Loki platform identification for the bus_loki reop.  My lack of a Loki makes it difficult for me to test.  Anybody with a Loki and dongle can update the firmware with `make upload` from the `bus_loki/rev_f` directory.  I haven't written the short Python script because I really don't know which repo and directory to stuff it in.

rohbotics, 2018-01-19:
Current status:

The script to read robot_ids has been merged into `ubiquity_motor`. It is called `probe_robot`.
@davecrawley is working on bumping the baud rate on the Magni side, there are some difficulties (getting checksum errors).
@waynegramlich has a PR https://github.com/UbiquityRobotics/bus_loki/pull/2 that needs to be tested and merged. (Wayne needs a Loki). 

Assuming the baud rate issues can be resolved, I think we are on track for wrapping this up by the end of January.

waynegramlich, 2018-01-27:
So far, no joy writing to the EEProm on Loki.  Normally, I'm pretty good with test points, but I forgot to add a few for the Loki EEPROM.

waynegramlich, 2018-01-28:
The "documentation" for the eeprom is at:

    https://www.raspberrypi.org/forums/viewtopic.php?t=108134

I purchased an Adafruit Perma-Proto Pi Hat with serial EEPROM.  My goal is to simply
read out the contents.

I run the following command to read the EEPROM:

    sudo ./eepflash.sh -r -d=1 -f=/tmp/foo.eep -t=24c32

and I get:

    Reading...
    dd: error reading '/sys/class/i2c-adapter/i2c-1/1-0050/eeprom': Connection timed out
    0+0 records in
    0+0 records out
    0 bytes copied, 0.0516922 s, 0.0 kB/s
    Error doing I/O operation.

It is pretty clear that my problem is currently a toolchain problem.  I'll noodle around on the net some more and see if I can figure out what is going wrong.

waynegramlich, 2018-02-07:
Everything has regressed.  I no longer get any indication that there is even a EEPROM on the bus.  I think it is time for me to solder some wires onto the appropriate pins and put it on a scope/logic analyzer. Sigh.

waynegramlich, 2018-02-14:
I still can't figure out my toolchain issue.  I put a scope on the SCL/SDA lines and ran `iodetect -y 1` and did not see any signals on the lines.  I'm stumped for now.  At this point in time, I think it is going to be easier to drop the Loki to use 38400 baud.

rohbotics, 2020-02-10:
Closing this since the Loki project isn't really being pursued anymore. We should have a way to do auto detect for any future Loki-like robot.

closed 2020-02-10 by rohbotics