UbiquityRobotics/ubiquity_main #89: Roscore as a seperate service

rohbotics, 2018-01-13:


rohbotics, 2018-01-15:
The service itself should be simple, source the environment and start roscore.

All the other ROS services should declare themselves `PartOf=roscore.service` and maybe `WantedBy=roscore.service` so that they get restarted with roscore.

rohbotics, 2018-01-15:
Idea for the roscore launch.

/etc/systemd/system/roscore.service 
```
[Unit]
After=NetworkManager.service time-sync.target
[Service]
Type=simple
User=ubuntu
ExecStart=/bin/sh -c "source /opt/ros/kinetic/setup.sh; source /etc/ubiquity/env.sh; roscore"
[Install]
WantedBy=multi-user.target
```

jim-v, 2018-01-15:
That seems reasonable

rohbotics, 2018-01-15:
It gets a little more complicated...

```
[Unit]
After=NetworkManager.service time-sync.target
[Service]
Type=forking
User=ubuntu
ExecStart=/bin/sh -c ". /opt/ros/kinetic/setup.sh; . /etc/ubiquity/env.sh; roscore & while ! echo exit | nc localhost 11311 > /dev/null; do sleep 1; done"
[Install]
WantedBy=multi-user.target
```

This forks to start roscore, and then the parent process only exits when the tcp port is opened successfully. Then because the Service type is forking, systemd adopts the orphaned process as the main process of the service.

rohbotics, 2018-01-15:
If magni-base is modified to look like the following it works

```
[Unit]
Requires=roscore.service
PartOf=roscore.service
After=NetworkManager.service time-sync.target roscore.service
[Service]
Type=simple
User=ubuntu
ExecStart=/usr/sbin/magni-base
[Install]
WantedBy=multi-user.target
```

closed 2018-01-15 by rohbotics