UbiquityRobotics/ubiquity_main #91: Go to Object Demo

rohbotics, 2018-01-20:


JanezCim, 2021-09-21:
I was assigned to review this but I dont know how to run it. Can @JanLipovsek please make a PR or something where there are instructions what exactly needs to get reviewed and with commands how to start it. Thanks :)

JanLipovsek, 2021-09-21:
These are all the repos that need to get reviewed:
    https://github.com/UbiquityRobotics/ur_vision_msgs.git
    https://github.com/UbiquityRobotics/detect_objects.git
    https://github.com/UbiquityRobotics/ur_object_tracking.git
    https://github.com/UbiquityRobotics/follow_me.git
    https://github.com/UbiquityRobotics/party_bot.git
Their usage is described in their corresponding readme.md files. The things to note here is that the code written to run on Noetic

closed 2021-10-11 by MoffKalast