UbiquityRobotics/ubiquity_main #93: Script Virtual Box Build

rohbotics, 2018-01-21:
Maybe share stuff with pi build.

Consider using ova/ovf to be vendor neutral.

jrlandau, 2018-01-21:
Sorry, I just don't understand what this is about.

Joe 408 455 44i4 if it's important enough, otherwise let's talk next
Saturday.

rohbotics, 2018-01-21:
Ideally we can automate the build process for the Virtual Box image, so that we can update it more easily. 

The Pi images are built using a script, maybe that can be extended to support building virtual machines as well.

jrlandau, 2018-01-21:
Sounds like a good idea--how would you suggest we proceed?

rohbotics, 2018-01-21:
@jrlandau We can discuss this next huddle.

For now this is not a high-priority issue, I just opened it as the next step since we did #29.