UbiquityRobotics/ubiquity_main #94: Can't install Gazebo sim

SystemDiagnosticss, 2018-01-25:
After command:
$ sudo apt-get install gazebo7
I got the following messages with some errors:

Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 gazebo7 : Depends: gazebo7-common (= 7.0.0+dfsg-2) but 7.9.0-1~xenial is to be installed
           Recommends: gazebo7-plugin-base but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

rohbotics, 2018-01-26:
What is the output of `apt policy gazebo7` ?

rohbotics, 2018-06-06:
Closing due to lack of information, re open if you still have the bug

closed 2018-06-06 by rohbotics