UbiquityRobotics/ubiquity_main #99: Bluetooth SDP RFCOMM problems

matealex, 2018-02-08:
I have a strange problem when trying the samples for RFCOMM server on a Raspberry Pi
If i run the script with python3 I can detect the service from other computer, but if i run it with python2.7 i can't detect it. 

"sudo sdptool browse local" outputs this:
With python 2.7:

> 
> Service Name: SampleServer
> Service RecHandle: 0x1000d
> Protocol Descriptor List:
>   "L2CAP" (0x0100)
>   "RFCOMM" (0x0003)
>     Channel: 1
> Profile Descriptor List:
>   "Serial Port" (0x1101)
>     Version: 0x0100

With python 3+:

> Service Name: SampleServer
> Service RecHandle: 0x1000d
> Service Class ID List:
>   UUID 128: 34b1cf4d-1069-4ad6-89b6-e161d79be4d8
>   "Serial Port" (0x1101)
> Protocol Descriptor List:
>   "L2CAP" (0x0100)
>   "RFCOMM" (0x0003)
>     Channel: 1
> Profile Descriptor List:
>   "Serial Port" (0x1101)
>     Version: 0x0100

I installed pybluez using these commands:

> sudo apt-get install libbluetooth-dev
> sudo apt-get install python-dev
> sudo pip install PyBluez
> sudo pip3 install PyBluez

What I want to do is have a ROS node that acts as a RFCOMM server and publishes messages to a topic. And since ROS runs the script with python2.7, my code doesn't work.

rohbotics, 2018-02-10:
@mjstn Can one of you have a look at this please?

rohbotics, 2018-07-27:
Closing due to age.
Please re-open if this is still an issue you face.

closed 2018-07-27 by rohbotics