UbiquityRobotics/ubiquity_main #111: New Kernel with 3B+ support

rohbotics, 2018-03-31:


rohbotics, 2018-04-14:
Works with version 20180313

pammelang, 2018-04-15:
where is this version? it's not the image on the link '2018-01-13-ubiquity-xenial-lxde' right? my rpi 3b+ isn't booting on that version.

rohbotics, 2018-04-15:
We haven't released an image with the new kernel yet, we hope to get that soon.

We need to do some more testing on the kernel/firmware combination to make sure that everything works and is stable, as well as get some other work finished before we can release a new image.

We hope to get a new image out before May that should have proper support for the Pi 3B+.

If you *really* want to try it now, you can if you are willing to jump through some hoops.
- Boot-up the latest image on a Pi 3 (not a Pi 3B+)
- Log in to the pi, edit the file `/etc/apt/sources.list.d/ubiquity-latest.list` to look like the following
```
deb https://packages.ubiquityrobotics.com/ubuntu/ubiquity xenial main

deb https://packages.ubiquityrobotics.com/ubuntu/ubiquity-testing xenial pi
```
- Run `sudo apt update; sudo apt upgrade`
- Shutdown the Pi 3, move the sd card into the Pi 3B+
- Boot up your Pi 3B+

If you chose to go this route, please post anything errors that you find in your `dmesg` logs, or any bugs with interfacing with hardware to this issue.

GuidoBartoli, 2018-05-31:
Hi, I'm trying to run `2018-01-13-ubiquity-xenial-lxde` on a **Raspberry Pi 3 B+** and it does not boot, too. Any update on this?

Thanks

rohbotics, 2018-06-04:
We are very close to a new image, if you would like to do some beta testing on it please contact me: ra@ubiquityrobotics.com

rohbotics, 2018-06-11:
Released and available, big thanks to the beta testers!
 
Image: 2018-06-07-ubiquity-xenial-lxde 
https://downloads.ubiquityrobotics.com/pi.html

closed 2018-06-11 by rohbotics