UbiquityRobotics/ubiquity_main #115: Raspberry Pi 3B+ Wifi

rohbotics, 2018-06-01:
There are issues with getting the wifi to work on the Pi 3B+, likely due to missing firmware.

rohbotics, 2018-06-02:
```
[    9.735579] brcmfmac: F1 signature read @0x18000000=0x15264345
[    9.742223] usbcore: registered new interface driver brcmfmac
[    9.792793] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.txt failed with error -2
```
 Relevant demsg lines

rohbotics, 2018-06-02:
Using firmware files from: https://github.com/RPi-Distro/firmware-nonfree/tree/master/brcm

Install brcmfmac43455-sdio.bin  brcmfmac43455-sdio.clm_blob  brcmfmac43455-sdio.txt to /lib/firmware/brcm 

dmesg after (looks happy)
```
[    4.636311] brcmfmac: F1 signature read @0x18000000=0x15264345
[    4.640154] usbcore: registered new interface driver brcmfmac
[    4.868591] random: crng init done
[    4.882371] brcmfmac: Firmware version = wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
[    4.882782] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 9.10.105 Compiler: 1.29.4 ClmImport: 1.36.3 Creation: 2018-03-09 18:56:28 
```

Wifi works, ap mode and 2.4 as well as 5g look happy

rohbotics, 2018-06-02:
@anfederman What was the USB issue you were talking about?

rohbotics, 2018-06-03:
This is now fixed.

closed 2018-06-03 by rohbotics

sajalyadav, 2019-05-29:
I'm having the same issue of wifi. How should I start wifi? How do I install these files. These files are already there in the mentioned folder. Do i need to only copy these files into my /lib/firmware/brcm.