UbiquityRobotics/ubiquity_main #125: Connect to PC no data on /joy topic on pc

droter, 2018-07-19:
Hi,

I have added my pc and raspberry pi IP address to the hosts file on both my PC and RPi.  I have added the ROS_MASTER and ROS_HOSTNAME_URI to both.  

If I ssh into the RPi and publish a topic from the command line I can see the data on my PC.  
RPi $ rostopic pub -r 10 /turtle1/cmd_vel geometry_msgs/Twist  '{linear:  {x: 0.1, y: 0.0, z: 0.0}, angular: {x: 0.0,y: 0.0,z: 0.0}}'

mypc $ rostopic echo /turtle1/cmd_vel
linear: 
  x: 0.1
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
---

If I try to echo the /joy topic from my pc no data shows up.

It looks like the node is being published on http://ubiquityrobot.local.  Could this be the issue?
$ rostopic info /joy
Type: sensor_msgs/Joy
Publishers: 
 * /joy_node (http://ubiquityrobot.local:40945/)

Thanks,

Matt

rohbotics, 2018-07-20:
Do you have a joystick connected? Which one?

There will be no joy messages being published if you don't have a Logitech Joystick plugged in.

droter, 2018-07-20:
Yes, I have the joystick plugged into the Raspberry Pi.

When I echo the topic from the Pi:
ubuntu@ubiquityrobot:~$ rostopic echo /cmd_vel
linear: 
  x: 0.75
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0

When I echo the topic from my pc:
$ rostopic echo /cmd_vel
(No Data)

I can see the /joy topic and /cmd_vel from the Raspberry Pi.  I can see the topics on my laptop but no data.  From answers.ros.org they suggest it is a network issue.

The only difference I can see:
One node is being published by /joy_node (http://ubiquityrobot.local:40945/)
And the other is being published by /rostopic_1666_1532041664121 (http://tractor:38879/)

I can see topic data from http://tractor but not http://ubuquityrobot.local.

rohbotics, 2018-07-22:
Sounds like there is a networking problem.

What is the output of `env | grep ROS` ?

droter, 2018-07-22:
On mypc:
$ env | grep ROS
ROS_ROOT=/opt/ros/kinetic/share/ros
ROS_PACKAGE_PATH=/home/matt/lino_ws/src:/home/matt/ros-a_ws/src:/opt/ros/kinetic/share
ROS_MASTER_URI=http://tractor:11311
ROS_VERSION=1
ROS_HOSTNAME=mypc
ROS_DISTRO=kinetic
ROS_ETC_DIR=/opt/ros/kinetic/etc/ros

on the Pi:
ubuntu@ubiquityrobot:~$ env | grep ROS
ROS_ROOT=/opt/ros/kinetic/share/ros
ROS_PACKAGE_PATH=/home/ubuntu/catkin_ws/src:/opt/ros/kinetic/share
ROS_MASTER_URI=http://tractor:11311
ROS_VERSION=1
ROS_PARALLEL_JOBS=-j2
ROS_HOSTNAME=tractor
ROSLISP_PACKAGE_DIRECTORIES=/home/ubuntu/catkin_ws/devel/share/common-lisp
ROS_DISTRO=kinetic
ROS_ETC_DIR=/opt/ros/kinetic/etc/ros

rohbotics, 2018-07-22:
Ahh ok, I see the problem.

Environment variables are shell-local on linux, not global. So when you are setting the ROS_MASTER_URI environment variable you are not changing for the service that runs it on boot.

Please edit /etc/ubiquity/env.sh, which is where the bootup launch process gets its environment configuration. This is also sourced into your shell by default on the Raspberry Pi.

Rohan

droter, 2018-07-22:
Yes! It works.

Very cool.  Thanks for your help.

Matt

closed 2018-07-22 by droter

rohbotics, 2018-07-22:
No problem. 

Good luck with your project!

In the future, questions like this are better suited for https://forum.ubiquityrobotics.com/, where they might get more visibility.