UbiquityRobotics/ubiquity_main #126: Ubiquity ros rpi image - bluetooth not working

ppedro74, 2018-07-22:
Bluetooth is not working and address showing up as AA:AA:AA:AA:AA:AA.
Is the Bluetooth supported ?

Thanks
 
Some info:
-----
#cat /proc/device-tree/model
Raspberry Pi 3 Model B Plus Rev 1.3
-----
#lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.5 LTS
Release:        16.04
Codename:       xenial
-----
#bluetoothd -v
5.41
-----
#hciconfig
hci0:   Type: Primary  Bus: UART
        BD Address: **AA:AA:AA:AA:AA:AA**  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING
        RX bytes:820 acl:0 sco:0 events:55 errors:0
        TX bytes:4632 acl:0 sco:0 commands:75 errors:0
-----
#sudo systemctl status bluetooth
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2018-07-22 23:15:39 UTC; 20min ago
     Docs: man:bluetoothd(8)
 Main PID: 18656 (bluetoothd)
   Status: "Running"
   CGroup: /system.slice/bluetooth.service
           └─18656 /usr/lib/bluetooth/bluetoothd

Jul 22 23:15:39 ubuntu systemd[1]: Starting Bluetooth service...
Jul 22 23:15:39 ubuntu bluetoothd[18656]: Bluetooth daemon 5.41
Jul 22 23:15:39 ubuntu systemd[1]: Started Bluetooth service.
Jul 22 23:15:39 ubuntu bluetoothd[18656]: Starting SDP server
Jul 22 23:15:39 ubuntu bluetoothd[18656]: Bluetooth management interface 1.14 initialized
Jul 22 23:15:39 ubuntu bluetoothd[18656]: Failed to obtain handles for "Service Changed" characteristic
Jul 22 23:15:39 ubuntu bluetoothd[18656]: Sap driver initialization failed.
Jul 22 23:15:39 ubuntu bluetoothd[18656]: sap-server: Operation not permitted (1)
Jul 22 23:15:41 ubuntu bluetoothd[18656]: Loading LTKs timed out for hci0

rohbotics, 2018-07-22:
Does it work if you remove the line `dtoverlay=pi3-miniuart-bt` from `/boot/config.txt` and reboot?

ppedro74, 2018-07-22:
I just commented and rebooted. Same result.

ppedro74, 2018-07-23:
still no success 

something changed no more: (Loading LTKs timed out for hci0 )

#hciconfig -a
hci0:   Type: Primary  Bus: UART
        BD Address: AA:AA:AA:AA:AA:AA  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING
        RX bytes:1606 acl:0 sco:0 events:57 errors:0
        TX bytes:2773 acl:0 sco:0 commands:57 errors:0
        Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
        Link policy: RSWITCH SNIFF
        Link mode: SLAVE ACCEPT
        Name: 'ubuntu'
        Class: 0x000000
        Service Classes: Unspecified
        Device Class: Miscellaneous,
        HCI Version: 4.1 (0x7)  Revision: 0x0
        LMP Version: 4.1 (0x7)  Subversion: 0x6119
        Manufacturer: Broadcom Corporation (15)
		
		
#sudo systemctl status bluetooth
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2018-07-22 23:56:28 UTC; 7min ago
     Docs: man:bluetoothd(8)
 Main PID: 783 (bluetoothd)
   Status: "Running"
   CGroup: /system.slice/bluetooth.service
           └─783 /usr/lib/bluetooth/bluetoothd

Jul 22 23:56:28 ubuntu systemd[1]: Starting Bluetooth service...
Jul 22 23:56:28 ubuntu bluetoothd[783]: Bluetooth daemon 5.41
Jul 22 23:56:28 ubuntu systemd[1]: Started Bluetooth service.
Jul 22 23:56:28 ubuntu bluetoothd[783]: Starting SDP server
Jul 22 23:56:28 ubuntu bluetoothd[783]: Bluetooth management interface 1.14 initialized
Jul 22 23:56:28 ubuntu bluetoothd[783]: Failed to obtain handles for "Service Changed" characteristic
Jul 22 23:56:28 ubuntu bluetoothd[783]: Sap driver initialization failed.
Jul 22 23:56:28 ubuntu bluetoothd[783]: sap-server: Operation not permitted (1)
		
#dmesg | egrep -i 'blue|firm'
[    0.159199] raspberrypi-firmware soc:firmware: Attached to firmware from 2018-03-13 18:52
[    4.990821] brcmfmac: Firmware version = wl0: Mar  1 2015 07:29:38 version 7.45.18 (r538002) FWID 01-6a2c8ad4
[    9.548237] Bluetooth: Core ver 2.22
[    9.548282] Bluetooth: HCI device and connection manager initialized
[    9.548295] Bluetooth: HCI socket layer initialized
[    9.548303] Bluetooth: L2CAP socket layer initialized
[    9.548319] Bluetooth: SCO socket layer initialized
[    9.565861] Bluetooth: HCI UART driver ver 2.3
[    9.565870] Bluetooth: HCI UART protocol H4 registered
[    9.565872] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    9.566035] Bluetooth: HCI UART protocol Broadcom registered
[    9.788795] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    9.788801] Bluetooth: BNEP filters: protocol multicast
[    9.788811] Bluetooth: BNEP socket layer initialized

ppedro74, 2018-07-23:
#ls /lib/firmware/brcm/*.hcd
does not show any bluetooth firmware driver

The Bluetooth **works** when used in a Raspberry **PI 3B (not the 3B+)** 

Fetch the Raspberry Pi 3B, Zero W Bluetooth firmware:
`wget -P /lib/firmware/brcm https://raw.githubusercontent.com/RPi-Distro/bluez-firmware/master/broadcom/BCM4345C0.hcd`

ppedro74, 2018-07-23:
After a few tests, I can get the bluetooth to work with **Raspberry PI 3B+** if the correct firmware is available.

Fetch the Raspberry Pi 3B+ Bluetooth firmware:
`wget -P /lib/firmware/brcm https://raw.githubusercontent.com/RPi-Distro/bluez-firmware/master/broadcom/BCM4345C0.hcd`

After the firmware is loaded, i can get the correct address. I can scan and pair a bluetooth LE gamepad.

@rohbotics Rohan:
You can close the thread,  I can help with more tests. 

Maybe makes sense to include both Bluetooth firmware files in the image distribution ?

alexisdal, 2018-08-13:
I did successfully downloaded the fimware

sudo wget -P /lib/firmware/brcm https://raw.githubusercontent.com/RPi-Distro/bluez-firmware/master/broadcom/BCM4345C0.hcd

Once downloaded, how do you enable the fw ? Starting the service fails.

```
root@ubiquityrobot:~# systemctl stop bluetooth.service
root@ubiquityrobot:~# systemctl start bluetooth.service
root@ubiquityrobot:~# systemctl status bluetooth.service
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
Condition: start condition failed at Mon 2018-08-13 13:01:02 CEST; 8s ago
           ConditionPathIsDirectory=/sys/class/bluetooth was not met
     Docs: man:bluetoothd(8)

Aug 13 13:00:13 ubiquityrobot systemd[1]: Stopped Bluetooth service.
Aug 13 13:00:54 ubiquityrobot systemd[1]: Stopped Bluetooth service.
```

and i see this in syslog

```
Aug 13 13:00:52 ubiquityrobot systemd[1]: hciuart.service: Service hold-off time over, scheduling restart.
Aug 13 13:00:52 ubiquityrobot systemd[1]: Stopped Configure Bluetooth Modems connected by UART.
Aug 13 13:00:52 ubiquityrobot systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Aug 13 13:00:54 ubiquityrobot systemd[1]: Stopped Bluetooth service.
Aug 13 13:01:02 ubiquityrobot hciattach[1943]: Initialization timed out.
Aug 13 13:01:02 ubiquityrobot hciattach[1943]: bcm43xx_init
Aug 13 13:01:02 ubiquityrobot systemd[1]: hciuart.service: Control process exited, code=exited status=1
Aug 13 13:01:02 ubiquityrobot systemd[1]: Failed to start Configure Bluetooth Modems connected by UART.
Aug 13 13:01:02 ubiquityrobot systemd[1]: hciuart.service: Unit entered failed state.
Aug 13 13:01:02 ubiquityrobot systemd[1]: hciuart.service: Failed with result 'exit-code'.
Aug 13 13:01:17 ubiquityrobot systemd[1]: hciuart.service: Service hold-off time over, scheduling restart.
Aug 13 13:01:17 ubiquityrobot systemd[1]: Stopped Configure Bluetooth Modems connected by UART.
Aug 13 13:01:17 ubiquityrobot systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Aug 13 13:01:27 ubiquityrobot hciattach[1951]: Initialization timed out.
Aug 13 13:01:27 ubiquityrobot hciattach[1951]: bcm43xx_init
Aug 13 13:01:27 ubiquityrobot systemd[1]: hciuart.service: Control process exited, code=exited status=1
Aug 13 13:01:27 ubiquityrobot systemd[1]: Failed to start Configure Bluetooth Modems connected by UART.
Aug 13 13:01:27 ubiquityrobot systemd[1]: hciuart.service: Unit entered failed state.
Aug 13 13:01:27 ubiquityrobot systemd[1]: hciuart.service: Failed with result 'exit-code'.

```

jonovos, 2018-10-14:
I too have downloaded loaded file **BCM4345C0.hcd** into folder **/lib/firmware/brcm** . Bluetooth communications still fail. ~~ I suspect there is some additional magical steps needed to tell the RPi3B+ to USE this new firmware. ... HOW?! How do we tell the Rpi to utilize this file? ... I have the same sad results as alexisdal does above.

ppedro74, 2018-10-15:
Hey Mark!

I made a blog post about this issue. Unfortunately my blog is down (migration in process).

So I'll burn a fresh Ubiquity image and explain all the steps.  Stay tuned!

ppedro74, 2018-10-15:
Let's start with a fresh Ubiquity image.

```
sudo hciconfig -a
ubuntu@ubiquityrobot:~$ 
```
Nothing shows up

```
sudo systemctl status hciuart
● hciuart.service - Configure Bluetooth Modems connected by UART
   Loaded: loaded (/lib/systemd/system/hciuart.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Mon 2018-10-15 16:19:41 UTC; 9s ago
  Process: 2162 ExecStart=/usr/bin/hciattach /dev/serial1 bcm43xx 921600 noflow - (code=exited, status=1/FAILUR

Oct 15 16:19:41 ubiquityrobot systemd[1]: hciuart.service: Control process exited, code=exited status=1
Oct 15 16:19:41 ubiquityrobot systemd[1]: Failed to start Configure Bluetooth Modems connected by UART.
Oct 15 16:19:41 ubiquityrobot systemd[1]: hciuart.service: Unit entered failed state.
Oct 15 16:19:41 ubiquityrobot systemd[1]: hciuart.service: Failed with result 'exit-code'.
```
The BT/Modem service is attached to serial1 and failed .

```
ls -lisa /dev/serial*
5635 0 lrwxrwxrwx 1 root root 7 Feb 11  2016 /dev/serial0 -> ttyAMA0
8473 0 lrwxrwxrwx 1 root root 5 Feb 11  2016 /dev/serial1 -> ttyS0
```
We have serial0 pointing to the hardware serial port (PL011) and the serial1 (BT Modem) is pointing to the mini UART.

```
sudo vi /opt/ros/kinetic/share/magni_bringup/param/base.yaml
ubiquity_motor:
  serial_port: "/dev/ttyAMA0"
  serial_baud: 38400
```
magni-base is connected to /dev/ttyAMA0 (serial 0). I recommend changing this to /dev/serial0 doing that works for PL011@serial0 miniuart@serial0.

If you are not using magni-base i recommend disabling the service.
`sudo systemctl disable magni-base.service`
You can keep it active, it can coexist with BT service.

[RaspberryPi overlays ](https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README)

> Switch Pi3 Bluetooth function to use the mini-UART (ttyS0) and restore
>         UART0/ttyAMA0 over GPIOs 14 & 15. Note that this may reduce the maximum
>         usable baudrate.
>         N.B. It is also necessary to edit /lib/systemd/system/hciuart.service
>         and replace ttyAMA0 with ttyS0, unless you have a system with udev rules
>         that create /dev/serial0 and /dev/serial1, in which case use
>         /dev/serial1 instead because it will always be correct. Furthermore,
>         you must also set core_freq=250 in config.txt or the miniuart will not
>         work.

Edit the config.txt and add the missing line: core_freq=250
```
sudo vi /boot/config.txt
# Additional overlays and parameters are documented /boot/overlays/README
dtoverlay=pi3-miniuart-bt
core_freq=250
dtoverlay=i2c-rtc,mcp7940x
```
although in the config.txt:
> core_freq
>      Frequency of GPU processor core in MHz. It has an impact on ARM
>     performance since it drives L2 cache.
> 
>      Default 250.
> 

[THE RASPBERRY PI UARTS](https://www.raspberrypi.org/documentation/configuration/uart.md)
[... 71 lines cut ...]
`sudo usermod -G bluetooth -a ubuntu`
user ubuntu added to bluetooth group.

***reboot***

```
bluetoothctl
[NEW] Controller 43:45:C0:00:3F:BC ubiquityrobot [default]

```

Conclusion:

1. BT connected to hardware port (PL011)
add the missing firmware file BCM4345C0.hcd
disable the pi3-miniuart-bt and don't add the core_freq=250

2. BT connected to the miniuart 
add the missing firmware file BCM4345C0.hcd
enable the overlay pi3-miniuart-bt add the core_freq=250

some users mentioned performance problems when using the core_freq=250, others mentioned 1/5 boots the Bluetooth is not correctly initialized .
So my recommendation is to use the hardware port with BT, and use USB/UART to connect to external devices. 

Mark: Let me know if it works.

rohbotics, 2018-10-15:
Thanks for the detailed analysis @ppedro74!

mjstn, 2018-10-15:
Yes, thanks for the details.   I am today going to try the above with the June 27 image.  I am familiar with all the commands you have discussed except was unclear that just doing the mget on the firmware with no other command would be enough for the firmware to be picked up.  I was trying last night some things based on comments in  https://hwblog.org/2018/06/08/how-to-get-bluetooth-working-on-raspberry-pi-3-model-b-or-b-with-arch-linux-arm but did not have success and in fact my hciconfig output stopped even showing up hci0 after that so I start with fresh image today and your post.  Your post with specifics you note are a great help to us.  Thank you for taking the time to reply in this organized way.   Cheers, Mark

closed 2018-11-17 by rohbotics

wegunterjr, 2020-01-04:
I am having such issues with uart comms.  This is giving me good info but I am still confused.
I want to use uart from pins 8,10 (14,15) to communicate with something.
I cannot figure out what I am missing.   
Thoughts?

davecrawley, 2020-01-04:
By default those pins are already in use for communicating with the Ubiquity Motor controller. Assuming you are using a ubiquity robotics Magni robot then you'll need to find another way to make your communication happen. Consider a USB-UART device and use the USB.