UbiquityRobotics/ubiquity_main #153: Wider Raspi connector and filled in pins

davecrawley, 2018-08-11:


rohbotics, 2018-08-25:
This is for preventing a 'sideways' offset while trying to plug the raspberry pi in.

mjstn, 2019-11-17:
Version 5.2 MCB is attempting to fix this.  We will evaluate if what was done was good enough as we evaluate version 5.2 MCB in late 2019

rohbotics, 2020-02-10:
@mjstn Would you call this done?

mjstn, 2020-02-10:
We have resolved it in a mediocre way but I have put numerous strong warnings in Learn pages as well.  So I feel this can be resolved 'best effort'.  The key thing is tall standoffs installed on MCB on both sides of tall RPi connector and those are on the BOM now.

rohbotics, 2020-02-10:
Ok, I will call this done then. Just trying to make sure that we don't have a bunch of junk floating around our backlog.

closed 2020-02-10 by rohbotics