UbiquityRobotics/ubiquity_main #159: Error installing image

xtrixpix, 2018-08-20:
I have a problem during installation of [ROS Kinetic on the Raspberry Pi](http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Kinetic%20on%20the%20Raspberry%20Pi) using the image and instructions from [here ](https://downloads.ubiquityrobotics.com/pi.html). After the image was restore when I tried to boot the first time Raspbbery I got the message _[  end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(179,2)_
Searching the internet gave me a solution: unplug any USB. This solution overcomes the above error but the boot stops at _[ end trace 93a58888580a7c63]_
Can someone help me?
Thank you.

rohbotics, 2018-08-21:
Which raspberry pi model do you have?

xtrixpix, 2018-08-21:
RaspberryPi 3B+

rohbotics, 2018-08-25:
Hm, not sure how to help. We have used this several times. Please verify the sha256 of the image, and try reflashing the card.

What are you using to flash the card?

xtrixpix, 2018-08-25:
What means flashing the card? What is sha256? Sorry about the questions but I don't understand.
Is it possible that ROS Kinetic are not compatible with RaspberryPi 3B+?

rohbotics, 2018-08-25:
We use this image on the RaspberryPi 3B+ frequently.

What operating system are you running on your laptop/desktop? Did you use GNOME Disk Tools?

Rohan

xtrixpix, 2018-08-25:
I have Ubuntu 16.04 (I think). About the program used to restor thw image I
just click on it.

rohbotics, 2018-08-25:
Please try restoring the SD card again.