UbiquityRobotics/ubiquity_main #163: usb camera not detected

rfsch, 2018-09-18:
hi 

I am unable to find a camera i have connected via USB using: 2018-06-27-ubiquity-xenial-lxde-raspberry-pi.img

please help

davecrawley, 2018-09-18:
Errr USB camera? The image is designed for the raspberry pi camera - which connects to the CSI port on the raspberry pi. The CSI port has a much higher data rate than is possible with most USB cameras, it is handled much more efficiently by the processor, has a higher resolution, much higher frame rate and in general will work way better than most USB cameras that you might use. I suggest you try a Raspberry Pi camera.

rfsch, 2018-09-19:
Hi Dave

thank you for your quick response. It is a special TOF camera by pmd (flexx). It has to be connected to the USB. It has a USB 3.0 cable. Maybe this is the problem ? How can I see whether the camera is detected ? I listed it using lsusb (where I can see its ID). I can see a driver for it in /dev/.
What looks strange is that in lsusb it says its "Unclaimed". Is there a menu in Lubuntu where I can select USB devices ? 

Thanks 

Rupert

rohbotics, 2018-09-19:
I have a couple questions.
What 'driver' to you see in /dev? If it is in /dev, what exactly in unrecognized?

What is the output of `dmesg`?

Rohan

rfsch, 2018-09-21:
Hi Rohan 

here it is: 

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.80-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1098 SMP Fri Mar 9 19:11:42 GMT 2018
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] cma: Reserved 8 MiB at 0x37800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 229376
[    0.000000] free_area_init_node: node 0, pgdat 80c6f800, node_mem_map b7016000
[    0.000000]   Normal zone: 2016 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 229376 pages, LIFO batch:31
[    0.000000] percpu: Embedded 14 pages/cpu @b6fcf000 s25600 r8192 d23552 u57344
[    0.000000] pcpu-alloc: s25600 r8192 d23552 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 227360
[    0.000000] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=1360 bcm2708_fb.fbheight=768 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 887580K/917504K available (7168K kernel code, 487K rwdata, 2032K rodata, 1024K init, 770K bss, 21732K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xb8800000 - 0xff800000   (1136 MB)
                   lowmem  : 0x80000000 - 0xb8000000   ( 896 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x80008000 - 0x80800000   (8160 kB)
                     .init : 0x80b00000 - 0x80c00000   (1024 kB)
                     .data : 0x80c00000 - 0x80c79cbc   ( 488 kB)
                      .bss : 0x80c7b000 - 0x80d3b9e4   ( 771 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] arm_arch_timer: Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000018] Switching to timer-based delay loop, resolution 52ns
[    0.000294] Console: colour dummy device 80x30
[    0.000310] console [tty1] enabled
[    0.000333] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000348] pid_max: default: 32768 minimum: 301
[    0.000653] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000663] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001661] Disabling memory control group subsystem
[    0.001734] CPU: Testing write buffer coherency: ok
[    0.001769] ftrace: allocating 22627 entries in 67 pages
[    0.049403] CPU0: update cpu_capacity 1024
[    0.049418] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.049453] Setting up static identity map for 0x100000 - 0x100034
[    0.051281] CPU1: update cpu_capacity 1024
[    0.051289] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.051933] CPU2: update cpu_capacity 1024
[    0.051940] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.052564] CPU3: update cpu_capacity 1024
[    0.052570] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.052659] Brought up 4 CPUs
[    0.052668] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.052672] CPU: All CPU(s) started in HYP mode.
[    0.052676] CPU: Virtualization extensions available.
[    0.053464] devtmpfs: initialized
[    0.065284] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.065569] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.065585] futex hash table entries: 1024 (order: 4, 65536 bytes)
[... 203 lines cut ...]
[   16.700126] Bluetooth: HCI UART driver ver 2.3
[   16.700133] Bluetooth: HCI UART protocol H4 registered
[   16.700135] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   16.700239] Bluetooth: HCI UART protocol Broadcom registered
[   16.850033] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   16.850039] Bluetooth: BNEP filters: protocol multicast
[   16.850049] Bluetooth: BNEP socket layer initialized
[   16.907392] IPv6: ADDRCONF(NETDEV_UP): enxb827eb6a35d1: link is not ready
[   17.589569] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   18.970695] Bluetooth: hci0 command 0x0c52 tx timeout
[   22.354846] fuse init (API version 7.26)
[... cut ...]
[   37.622965] input: RAPOO Rapoo Touch Mouse as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.2/1-1.1.2:1.1/0003:24AE:1801.0004/input/input5
[   37.691850] hid-generic 0003:24AE:1801.0004: input,hiddev0,hidraw3: USB HID v1.10 Mouse [RAPOO Rapoo Touch Mouse] on usb-3f980000.usb-1.1.2/input1
[   53.780192] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   53.809959] ip_tables: (C) 2000-2006 Netfilter Core Team
[   53.827040] nf_conntrack version 0.5.0 (14336 buckets, 57344 max)

Rupert

rfsch, 2018-09-21:
Hi Rohan 

and here the ROS master.log

[rosmaster.main][INFO] 2018-09-18 18:45:14,542: initialization complete, waiting for shutdown
[rosmaster.main][INFO] 2018-09-18 18:45:14,543: Starting ROS Master Node
[xmlrpc][INFO] 2018-09-18 18:45:14,545: XML-RPC server binding to 0.0.0.0:11311
[xmlrpc][INFO] 2018-09-18 18:45:14,546: Started XML-RPC server [http://ubiquityrobot.local:11311/]
[rosmaster.master][INFO] 2018-09-18 18:45:14,546: Master initialized: port[11311], uri[http://ubiquityrobot.local:11311/]
[xmlrpc][INFO] 2018-09-18 18:45:14,547: xml rpc node: starting XML-RPC server
[rosmaster.master][INFO] 2018-09-18 18:45:14,588: +PARAM [/run_id] by /roslaunch
[rosmaster.master][INFO] 2018-09-18 18:45:14,593: +PARAM [/roslaunch/uris/host_ubiquityrobot_local__42721] by /roslaunch
[rosmaster.master][INFO] 2018-09-18 18:45:14,995: +PARAM [/rosversion] by /roslaunch
[rosmaster.master][INFO] 2018-09-18 18:45:14,996: +PARAM [/rosdistro] by /roslaunch
[rosmaster.master][INFO] 2018-09-18 18:45:15,288: +SERVICE [/rosout/get_loggers] /rosout http://ubiquityrobot.local:34273/
[rosmaster.master][INFO] 2018-09-18 18:45:15,292: +SERVICE [/rosout/set_logger_level] /rosout http://ubiquityrobot.local:34273/
[rosmaster.master][INFO] 2018-09-18 18:45:15,298: +PUB [/rosout_agg] /rosout http://ubiquityrobot.local:34273/
[rosmaster.master][INFO] 2018-09-18 18:45:15,311: +SUB [/rosout] /rosout http://ubiquityrobot.local:34273/
[rosmaster.master][INFO] 2018-09-19 19:11:18,549: +PARAM [/roslaunch/uris/host_ubiquityrobot_local__33405] by /roslaunch
[rosmaster.master][INFO] 2018-09-19 19:11:18,561: +PARAM [/rosversion] by /roslaunch
[rosmaster.master][INFO] 2018-09-19 19:11:18,562: +PARAM [/rosdistro] by /roslaunch
[rosmaster.master][INFO] 2018-09-19 19:11:18,563: +PARAM [/royale_camera_driver/node_name] by /roslaunch
[rosmaster.master][INFO] 2018-09-19 19:11:18,893: +PUB [/rosout] /royale_camera_static_tf http://ubiquityrobot.local:39139/
[rosmaster.master][INFO] 2018-09-19 19:11:18,896: publisherUpdate[/rosout] -> http://ubiquityrobot.local:34273/ ['http://ubiquityrobot.local:39139/']
[rosmaster.master][INFO] 2018-09-19 19:11:18,900: +SERVICE [/royale_camera_static_tf/get_loggers] /royale_camera_static_tf http://ubiquityrobot.local:39139/
[rosmaster.master][INFO] 2018-09-19 19:11:18,904: publisherUpdate[/rosout] -> http://ubiquityrobot.local:34273/ ['http://ubiquityrobot.local:39139/']: sec=0.01, result=[1, '', 0]
[rosmaster.master][INFO] 2018-09-19 19:11:18,905: +SERVICE [/royale_camera_static_tf/set_logger_level] /royale_camera_static_tf http://ubiquityrobot.local:39139/
[rosmaster.master][INFO] 2018-09-19 19:11:18,914: +PUB [/tf] /royale_camera_static_tf http://ubiquityrobot.local:39139/
[rosmaster.master][INFO] 2018-09-19 19:11:19,039: +PUB [/rosout] /royale_camera_driver http://ubiquityrobot.local:41169/
[rosmaster.master][INFO] 2018-09-19 19:11:19,043: +SERVICE [/royale_camera_driver/get_loggers] /royale_camera_driver http://ubiquityrobot.local:41169/
[rosmaster.master][INFO] 2018-09-19 19:11:19,046: +SERVICE [/royale_camera_driver/set_logger_level] /royale_camera_driver http://ubiquityrobot.local:41169/
[rosmaster.master][INFO] 2018-09-19 19:11:19,055: +PARAM [/royale_camera_driver] by /royale_camera_driver
[rosmaster.master][INFO] 2018-09-19 19:11:19,096: publisherUpdate[/rosout] -> http://ubiquityrobot.local:34273/ ['http://ubiquityrobot.local:39139/', 'http://ubiquityrobot.local:41169/']
[rosmaster.master][INFO] 2018-09-19 19:11:19,099: publisherUpdate[/rosout] -> http://ubiquityrobot.local:34273/ ['http://ubiquityrobot.local:39139/', 'http://ubiquityrobot.local:41169/']: sec=0.00, result=[1, '', 0]
[rosmaster.master][INFO] 2018-09-19 19:11:19,140: +PUB [/rosout] /royale_camera http://ubiquityrobot.local:40415/
[rosmaster.master][INFO] 2018-09-19 19:11:19,145: +SERVICE [/royale_camera/get_loggers] /royale_camera http://ubiquityrobot.local:40415/
[rosmaster.master][INFO] 2018-09-19 19:11:19,148: +SERVICE [/royale_camera/set_logger_level] /royale_camera http://ubiquityrobot.local:40415/
[rosmaster.master][INFO] 2018-09-19 19:11:19,157: +SERVICE [/royale_camera/load_nodelet] /royale_camera http://ubiquityrobot.local:40415/
[rosmaster.master][INFO] 2018-09-19 19:11:19,160: +SERVICE [/royale_camera/unload_nodelet] /royale_camera http://ubiquityrobot.local:40415/
[rosmaster.master][INFO] 2018-09-19 19:11:19,162: +SERVICE [/royale_camera/list] /royale_camera http://ubiquityrobot.local:40415/
[rosmaster.master][INFO] 2018-09-19 19:11:19,197: publisherUpdate[/rosout] -> http://ubiquityrobot.local:34273/ ['http://ubiquityrobot.local:39139/', 'http://ubiquityrobot.local:41169/', 'http://ubiquityrobot.local:40415/']
[rosmaster.master][INFO] 2018-09-19 19:11:19,203: publisherUpdate[/rosout] -> http://ubiquityrobot.local:34273/ ['http://ubiquityrobot.local:39139/', 'http://ubiquityrobot.local:41169/', 'http://ubiquityrobot.local:40415/']: sec=0.00, result=[1, '', 0]
[rosmaster.master][INFO] 2018-09-19 19:11:20,407: -PUB [/rosout] /royale_camera_driver http://ubiquityrobot.local:41169/
[rosmaster.master][INFO] 2018-09-19 19:11:20,408: publisherUpdate[/rosout] -> http://ubiquityrobot.local:34273/ ['http://ubiquityrobot.local:39139/', 'http://ubiquityrobot.local:40415/']
[rosmaster.master][INFO] 2018-09-19 19:11:20,410: publisherUpdate[/rosout] -> http://ubiquityrobot.local:34273/ ['http://ubiquityrobot.local:39139/', 'http://ubiquityrobot.local:40415/']: sec=0.00, result=[1, '', 0]
[... cut ...]
[rosmaster.master][INFO] 2018-09-19 19:11:20,758: +PARAM [/royale_camera_driver] by /royale_camera_driver
[rosmaster.master][INFO] 2018-09-19 19:11:20,804: publisherUpdate[/rosout] -> http://ubiquityrobot.local:34273/ ['http://ubiquityrobot.local:39139/', 'http://ubiquityrobot.local:40415/', 'http://ubiquityrobot.local:45445/']
[rosmaster.master][INFO] 2018-09-19 19:11:20,807: publisherUpdate[/rosout] -> http://ubiquityrobot.local:34273/ ['http://ubiquityrobot.local:39139/', 'http://ubiquityrobot.local:40415/', 'http://ubiquityrobot.local:45445/']: sec=0.00, result=[1, '', 0]

rohbotics, 2018-09-22:
Looks like the device is being properly detected:

```
[ 2.410678] usb 1-1.2: new high-speed USB device number 5 using dwc_otg
[ 2.541414] usb 1-1.2: New USB device found, idVendor=1c28, idProduct=c012
[ 2.541424] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2.541432] usb 1-1.2: Product: FX3
[ 2.541440] usb 1-1.2: Manufacturer: PMDtec
```

Can you post the output of `lsusb -v` and `ls -al /dev` as well?

This looks like there is no driver available for the USB device, in which case you will need to find how to install that. Another thing to check are if there are any udev rules or anything that the manufacturer tells you to install. Other than that there is little we can really do to support you.

rfsch, 2018-09-22:
Hi Rohan 

here we go: 

lsusb -v 

      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      2 TT per port
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 Unused
  bDeviceProtocol         1 Single TT
  bMaxPacketSize0        64
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0002 2.0 root hub
  bcdDevice            4.09
  iManufacturer           3
  iProduct                2
  iSerial                 1
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           25
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
[... 926 lines cut ...]

# f1 Amundsen
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1c28", ATTRS{idProduct}=="c02e", MODE="0666", SYMLINK+="f1-%k", GROUP="plugdev"

# m1 Amundsen
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1c28", ATTRS{idProduct}=="c02f", MODE="0666", SYMLINK+="m1-%k", GROUP="plugdev"
============================================================
driver manufacturer README

This directory contains udev-rules files.
See https://wiki.debian.org/udev for explanation how udev rules work.

If you are using ubuntu, copy the .rules file to /etc/udev/rules.d/ to use pmd devices without administrative privileges (you will need root rights once to copy the file to the destination):

sudo cp 10-royale-ubuntu.rules /etc/udev/rules.d

Please make sure that the user is in the group plugdev!

After copying the udev rule, please unplug and reinsert the camera.
================================================================
"getent group plugdev" shows 

plugdev:x:46:ubuntu

Thank you so much looking into this.

rfsch, 2018-09-22:
Dear Rohan

thanks for your effort and time!

Cheers

Rupert