UbiquityRobotics/ubiquity_main #164: Problem using GPIO

Leonardo092, 2018-11-08:
Hello there,
I downloaded the image of Ubuntu Mate from your [web page](https://downloads.ubiquityrobotics.com/pi.html) and made it run on Raspberry Pi3 B+. I am using GPIO to send commands through the board but, when connecting to GPIO6, the system suddenly resets. Is there some script/service running in background that relies on that specific GPIO?
I already disabled **roscore.service** and **magni-base**.

rohbotics, 2018-11-08:
The buttons and leds device tree file might be grabbing those pins.

Check the bottom of /boot/config.txt and comment  that dtbo out.

I'm mobile right now, I can provide more details later if needed.

Leonardo092, 2018-12-03:
Thanks, you solved the problem! Sorry for answering that late.

closed 2018-12-03 by Leonardo092