UbiquityRobotics/ubiquity_main #189: Uncontrolled motor movement after killing motor node

davecrawley, 2019-03-05:
Start magni motor node
Publish a twist message periodically on /cmd_vel (e.g., 0.1m/s forward, 30Hz)
Observe Magni moves forward as expected
Stop motor node
Observe Magni stops as expected
Stop publishing messages on /cmd_vel
Start magni motor_node
Observe Magni is moving at max velocity for a short period of time seemingly proportional to the original velocity.
This looks like the firmware’s integral is winding up when it is in an unconnected state.
Since we regard this kind of uncontrolled movement as a security issue, we would greatly appreciate help with it.

mjstn, 2019-11-17:
This has been fixed in hardware from MCB 5.x versions combined with firmware fixes for ESTOP and PID related code.   This was first addressed in firmware version v32 and was later emproved upon in v35 firmware.    I believe this to be resolved.

rohbotics, 2020-07-09:
Closing this due to age, and what we think is a solution with a lack of reproduction since.

closed 2020-07-09 by rohbotics