UbiquityRobotics/ubiquity_main #199: Wifi access point

LucasLabarussiat, 2019-06-18:
Hello,

I installed your image found [here](https://downloads.ubiquityrobotics.com/pi.html) on my Raspberry Model 3B+.
At the beginning I know I saw the wifi access point (SSID ubiquityrobotXXXX). At the time I didn't need it, I connected to my wifi network and ran "sudo systemctl disable magni-base" in a terminal.

However, I now have a use for it, so I checked but it seems the SSID doesn't run anymore. I can't see it from my computer. I tried running "sudo systemctl enable magni-base" and it worked, but after a reboot no sign of the wifi access point.

I am really new both to UNIX and Raspberry, so this isn't easy... Could you please tell me how to make this wifi access point available again?

Thank you

rohbotics, 2019-06-18:
Hi @LucasLabarussiat,

The wifi access point is provided by [pifi](https://github.com/rohbotics/pifi), and will automatically start if there are no wifi networks found at boot. 

Enabling and disabling magni-base doesn't change pifi, they only start and stop the ROS services used by [Magni](https://ubiquityrobotics.com/products/). I would recommend keeping it disabled if you are not using a magni.

Thanks for your question, and have fun with our image.
Feel free to open more issues, or use our https://forum.ubiquityrobotics.com/ if you have further questions.

LucasLabarussiat, 2019-06-19:
Hello Rohan,

Thank you for your very quick answer. So I just deleted the additional wifi
networks I saved, and it is working fine. I feel quite stupid for not
thinking about it myself haha.

Thank you so much,

*Lucas LABARUSSIAT*
------------------------------
ÉCOLE DES MINES D'ALES                             FIGEAC AÉRO
<http://www.figeac-aero.com/>
Promotion Mécatronique 3                Apprenti ingénieur R&D
------------------------------

rohbotics, 2019-06-19:
Haha no problem, glad we figured it out.

closed 2019-06-19 by rohbotics