UbiquityRobotics/ubiquity_main #205: Raspberry Pi time setting

zhanghaijason, 2019-07-15:
Hello,
I tried to set the system time correctly yesterday and did it. However, today when it boots up. The time went back to 9:30, Feb 11st, 2016 again. Is there a way to set the time correctly? I tried to use internet to set time automatically but it didn't work. Thanks.

rohbotics, 2019-07-17:
Are you using a Magni? Or just a Raspberry Pi? If you are using just a Pi, do you have an RTC connected?

zhanghaijason, 2019-07-17:
> Are you using a Magni? Or just a Raspberry Pi? If you are using just a Pi, do you have an RTC connected?

Hi Just a Pi. And I don't have an RTC. Is purchaing a RTC the only solution? Thanks.

davecrawley, 2019-07-17:
The Raspberry pi doesn't have a real time clock on board, what that means is that when the RPi shuts down it forgets the time and goes back to a default setting. We put a real time clock on our Magni robot for this reason - so if you get a Magni this works correctly. 

Its not a complete solution but perhaps you could consider writing a short script that stores the latest clock information in the file system, and then another that reads that clock information on boot up. That way at least the clock will not be too out of date. Usually its when the clock is several years out of date that you start having serious problems in ROS message passing, so this might prevent some of the most fatal bugs.

There are usually lots of clock sources available, obviously if its connected to the internet then NTP servers are available, but even if its on a local network, most of the computers on that network have time on them so at least theoretically you can get the time from them.

As I mentioned we don't need it for our Magni product - however if you were to write a set of scripts and open source them, we'd likely put it on the default image as long as it passes our technical review and it is open-sourced with a permissive license (e.g. a BSD license). We'd put it on the image for the benefit of the community.

Let me know if this is interesting for you.

zhanghaijason, 2019-07-17:
> The Raspberry pi doesn't have a real time clock on board, what that means is that when the RPi shuts down it forgets the time and goes back to a default setting. We put a real time clock on our [...]

Thanks for the information. Since there is no easy way to do this I would set time every time manually. I have another question. I tried to install opencv and followed this [https://www.pyimagesearch.com/2017/09/04/raspbian-stretch-install-opencv-3-python-on-your-raspberry-pi/](url).  it asked me to `Open your /etc/dphys-swapfile  and then edit the CONF_SWAPSIZE  variable to 1024 `. However, I didn't find such file. So I guess the default CONF_SWAPSIZE is 100. This "Out of memory" error caused my compliation fail. Is there a way to fix this?  
Or did you try install opencv before? Thanks.

mjstn, 2019-07-17:
All of this is to help you and is not part of Magni support.
What I will add is these two things:
1) RTC without a Magni is often done using a tiny plug-in board on the RasPi connector.
Look into this one https://www.adafruit.com/product/3386    
There are even cheaper solutions but Adafruit is great for support of their products so suggest that.

2) If you do not have /etc/dphys-swapfile  you have not installed swap.
Please look into help for install of   dphys-swapfile  which will use sudo apt-get install dphys-swapfile

If this has been helpful and because this it not a Magni issue, please consider optionally 'buying us a cup of coffee' from the donate button on pi image download:  https://downloads.ubiquityrobotics.com/pi.html

Good luck in your efforts.

mjstn, 2019-07-17:
In looking at my notes I realize AdaFruit may be explaining things for the Raspbian image.
What I had done on a prior robotics project that used Ubuntu 16.04 was more like this:
https://ubuntu-mate.community/t/supporting-ds3231-rtc-source-synchronise-with-network-time/5711
It is a little tricky so research looking for DS3231 boards that are also cheap for a Pi and also look for help from persons posting projects using the DS3231 board on a Raspberry Pi running ubuntu 16.04 and that will help you  but will take some effort.

rohbotics, 2019-07-19:
If you do end up buying an RTC, you can simply edit `/boot/config.txt` tas described here https://github.com/UbiquityRobotics/pi_image_build#rtc

rohbotics, 2019-07-22:
Hi @zhanghaijason I am closing this because there is not much that be done to resolve this issue. Feel free to ask any remaining questions you may have.

closed 2019-07-22 by rohbotics