UbiquityRobotics/ubiquity_main #209: Unable to login to my Raspberry

samurairepo, 2019-10-02:
Hello,
please I tried to follow the instructions on https://downloads.ubiquityrobotics.com/pi.html
I was able to run it and I am the login prompt.
I can enter as Guest, but I have limited rights (sudo adduser XXX - asks for guest password)
What is the sudo password ? How do I create new user ?

rohbotics, 2019-10-02:
Instead of logging in as Guest, log in as 'Ubuntu user' and use the password `ubuntu`

rohbotics, 2019-10-02:
@samurairepo Just noticed your edit, don't worry about it, we all have those days.

Good luck with whatever you are using our image for! Please consider "buying us a coffee" via the donate button on the downloads page or Paypal to tips@ubiquityrobotics.com.

rohbotics, 2020-02-10:
Closing this because it is resolved.

closed 2020-02-10 by rohbotics