UbiquityRobotics/ubiquity_main #221: Download kernel headers

h3ct0r, 2020-02-06:
Hi, is it possible to download the kernel headers for the latest version of the Raspberry Pi image? 

The image is currently using the "4.14.98-v7+" kernel, but I cannot find any place to download the kernel-headers, and there are not installed by default or in the repos.

Thanks!

rohbotics, 2020-02-06:
Thanks for the question.

You should be able to `sudo apt install raspberrypi-kernel-headers` let me know if that doesn't work.

h3ct0r, 2020-02-06:
Hi @rohbotics thanks for the answer, but even after installing the `raspberrypi-kernel-headers` the system does not have the `/lib/modules/4.14.98-v7+/build` or `/lib/modules/4.14.98-v7+/source` folders.

Even `sudo apt-get install linux-headers` results in a `E: Package 'linux-headers' has no installation candidate` message. 

Do you have any other idea?

rohbotics, 2020-02-06:
Are you sure you are running the latest kernel? I get `4.19.50-v7+` from `uname -r`
What is the output of `apt policy raspberrypi-kernel`? 

I see `/lib/modules/4.19.50-v7+/build` after installing the headers/

h3ct0r, 2020-02-07:
Hi, @rohbotics; thank you! I verified that I wasn't running the latest kernel as you said. After performing an `apt update && apt upgrade` and an `apt install raspberrypi-kernel-headers`, everything is running as expected. I needed to update all system packages!

closed 2020-02-07 by h3ct0r

rohbotics, 2020-02-07:
Thanks for closing the issue. Good luck with your project!