UbiquityRobotics/ubiquity_main #235: Raspberry Pi 4 B refuses SSH connection

kristjanp28, 2020-03-05:
Refuses SSH connection in AP mode or connected to the network. Tried creating SSH file in BOOT partition.

karpandrew12, 2020-03-21:
Im having the exact issue. Im trying to ssh into my raspberry pi 4 from my ubuntu machine and its stating the password is incorrect.

karpandrew12, 2020-03-21:
I ended up figuring out how to fix this. Before i tried ssh from development computer to robot comp with ssh pi@ipaddress. This was incorrect. The correct way is to ssh the name of your system in replacement to "pi". For example : ssh ubuntu@ipaddress.

I hope this helps.

kristjanp28, 2020-03-21:
> I ended up figuring out how to fix this. Before i tried ssh from development computer to robot comp with ssh pi@ipaddress. This was incorrect. The correct way is to ssh the name of your system in replacement to "pi". For example : ssh ubuntu@ipaddress.
> 
> I hope this helps.

There is no issue with wrong password. There is an issue with Pi itself or with the network that we are using.

rohbotics, 2020-07-09:
Closing this because we found out that the issue was the ssh systemd service being configured badly.

closed 2020-07-09 by rohbotics

an-dr, 2020-07-09:
@rohbotics should we update the image? Or could you give some hint hot to fix it on our existing image? I have the same issue, but my fix is to wait about 5-10 minutes :)

rohbotics, 2020-07-09:
@an-dr The fix is here: https://forum.ubiquityrobotics.com/t/ros-image-on-raspberry-pi-4/326/59?u=rohbotics

It will be in the next image once we release that.