UbiquityRobotics/ubiquity_main #238: upgrade_firmware.py  wrong access token error error give no hint at what is wrong

mjstn, 2020-03-14:
When an incorrect access token is entered in upgrade_firmware.py the script accepts it which leads the user to think the code was ok.
Then a version is entered and right away this error comes up

Error downloading firmware 401

A real geek may realize this is the 'Unauthorized' error but it is going to lead to confusion if we do not explain it in human readable form.

To help users move on we should help by saying it is error 401 like above but also then say  
'unauthorized access most likely due to incorrect or timed out token entry'.
The access tokens are only good for one hour from the time of reception.

rohbotics, 2020-03-14:
Good catch. Right now we are just printing out the error codes, but it would be very easy to interpret some common ones (401, 404, and maybe the 500 series).

rohbotics, 2020-03-20:
@mjstn https://github.com/UbiquityRobotics/ubiquity_motor/pull/81 should address this.

closed 2020-03-28 by rohbotics