UbiquityRobotics/ubiquity_main #240: Image 2020-02-10-ubiquity-xenial-lxde Compressed Data Is Corrupt

esduran, 2020-04-03:
Hello,

I downloaded the latest image from your website on Mac OS 10.15.4 and when I execute the following command:

`xz -d image_name.img.xz`

I receive the following error:

`Compressed data is corrupt`

It all works fine with image https://ubiquity-pi-image.sfo2.cdn.digitaloceanspaces.com/2019-06-19-ubiquity-xenial-lxde-raspberry-pi.img.xz

esduran, 2020-04-03:
On Mac OS there is no sha256sum so I used `shasum -a 256 image_name.img.xz` and ended up getting **31c9813644010b4bafc82105627a8941db957b53be6c2b6b506e4283cbc0f6e6** which differs form the one on the website.

rohbotics, 2020-04-04:
I get the same checksum when I download it, sounds like something got corrupted on the way to you.

esduran, 2020-04-05:
Might be an Apple thing. I tried downloading it through my iPhone and I get the same checksum that my Mac is getting. I'll try on an Ubuntu VM.

rohbotics, 2020-07-09:
Closing due to age and no recent complaints.

closed 2020-07-09 by rohbotics