UbiquityRobotics/ubiquity_main #242: Evaluate State Machine Libraries

rohbotics, 2020-07-10:
We should evaluate all the available state machine libraries for ROS, C++ and Python.

Some ones off the top of my head
- ROS SMACC
- Boost::SML
- ROS BehaviorTrees

DOD: Some comparison document somewhere.

rohbotics, 2020-07-13:
@kristjanreba please assign yourself to this issue when you accept the UbiquityRobotics org invite.

kristjanreba, 2020-07-14:
Comparing different state machine libraries for ROS.

**ROS SMACC** https://discourse.ros.org/t/introducing-the-smacc-state-machine-library/14963
\+ fast
\+ relatively easy to use
\- only C++
\- new (no big community yet)

**Boost::SML** https://github.com/boost-ext/sml
\+ large community
\- only C++

**ROS BehaviourTrees**
\+ light weight
\+ C++ and Python
\- not much community behind the project
\- a bit too complicated

**SMACH** http://wiki.ros.org/smach
\+ quick to implement in python
\+ high level and easy to understand
\+ larger community
\- slow on low level systems
\- only Python

Those are my impressions of the libraries.
I would use the SMACH for python code and ROS SMACC or Boost::SML for C++ (low level) code.

rohbotics, 2020-07-14:
@kristjanreba Thanks for the comparison.

closed 2020-07-14 by rohbotics