UbiquityRobotics/ubiquity_main #243: Problem with Pi4 install

jacepang, 2020-08-04:
When copying the image to Pi 4B (8G), it shows that stat4x.elf is not compatible 0x0.
Is it not support the pi4 yet ?

MoffKalast, 2020-08-04:
Hmm well we've only ever tested Pi 4s with 1,2 and 4G of memory, so it might be that it's causing some unforeseen issues. I recall Raspbian having to push out a new update to support it so we'll likely have to do the same.

benjamintanweihao, 2020-10-22:
Can confirm that it doesn't work on a Raspi 4 with 8GB. Is there anything we can do to attempt a fix?

davecrawley, 2020-10-22:
Currently the distribution uses 32 bit linux and ROS. Obviously a 32 bit
machine is limited to 4Gb machines.

We are working on a 64 bit distribution  - however a critical driver (i.e.
the MMAL camera driver) does not work on 64 bit yet and currently the
maintainers of those packages are not receptive to our request to make them
work. We are actively working on work arounds for these issues, however
currently there is a significant block on making the whole system work
right with a 64bit system, and without a 64bit system you can't go above
4Gb.

If you are very motivated to work on a 8Gb machine getting a 64bit version
of the MMAL driver working would accelerate our process and we would
definitely collaborate to quickly release a distribution if this issue can
be solved. I expect that we will find a way to address it eventually, but
it may take some time.

For more on the MMAL driver:
https://www.raspberrypi.org/documentation/raspbian/applications/camera.md

David

benjamintanweihao, 2020-10-22:
Thanks for the super quick reply! Somehow I managed to get the image to work with the follolwing:

I downloaded `https://github.com/raspberrypi/firmware/tree/stable/boot` and replaced all the `*.elf` and `*.dat` files in the boot and after that I managed to SSH!

rohbotics, 2020-10-22:
Sorry this fell through the cracks. 

The 8GB model does support booting in 32 bit mode, and you can address all 8GB of RAM, but any single process will be limited to 3GB of memory. This is a limitation of the linux kernel's PAE (Physical Address Extension) system.

I am making a test image for you that has the newer kernel, it should be ready in a couple hours. It would be pretty helpful if you could test that out to make sure it works on the 8GB model. The change is similar to what you have done with manually replacing the kernel image, but this also updates the kernel modules, firmware, and device tree, so you are less likely to run into subtle issues.

davecrawley, 2020-10-22:
Cool!

Are you seeing the full memory space? Or is it only addressing the first 4
Gb?

Thanks

David

benjamintanweihao, 2020-10-22:
@rohbotics  No worries! I'm glad that I finally got it to work and super impressed that you guys are so quick at responding. <3 <3 <3. I'd be happy to test the new image out. 

@davecrawley Is there a way to check? I've tried `htop` and it says `7.73G` so great success?

rohbotics, 2020-10-22:
@benjamintanweihao Here is the image for you to test out, please let me know how it goes. https://ubiquity-pi-image.sfo2.cdn.digitaloceanspaces.com/2020-10-22-ubiquity-xenial-lxde-rpi-testing.img.xz

benjamintanweihao, 2020-10-23:
@rohbotics It works! I was able to flash the image on a 64GB SD card and the Raspi 4 booted successfully. `htop` shows all the RAM and also I could execute the `raspicam_node` successfully :D. 

Great job! Anything else I should check for?

rohbotics, 2020-10-26:
Those are the 2 big ones.

If you are planning on continuing to use that image, I would switch it out of testing mode, so that our beta updates don't break you on apt-get upgrades.

Edit `/etc/apt/sources.list.d/ubiquity-latest.list` and change every `https://packages.ubiquityrobotics.com/ubuntu/ubiquity-testing` to `https://packages.ubiquityrobotics.com/ubuntu/ubiquity`.

benjamintanweihao, 2020-10-26:
Copy that! Thanks!

closed 2021-08-17 by rohbotics