UbiquityRobotics/ubiquity_main #258: LS lidar driver investigations

MoffKalast, 2022-01-11:
Hard to say what but there is something weird with the timestamps and data coming out of the LS lidar driver, and it needs to be looked into.

Anecdotal evidence:
- avalon claims the sick lidar outperforms the ls lidar despite obvious hardware shortcomings
- odd occurences where lama cannot provide a transform
- odd occurences where lidar data cannot be transformed into /map frame
- lidar timings sometimes being set a few ms in the future

anfederman, 2022-01-11:
I am uncertain why data coming over ethernet would be out of sync with data
over USB. Don't they both get
processed in the same hardware chip?  Are there some sort of filters or
avengers that sample the Lidar data (say over 3 measurements)
and then 'predict' what the signal should be?

MoffKalast, 2022-01-11:
I'm not sure where the time stamps are made, that's why this issue is here to find out.