UbiquityRobotics/ubiquity_main #259: Focal image: things left to fix

MoffKalast, 2022-01-17:


MoffKalast, 2022-02-02:
Question, @JanezCim does pifi now work normally on focal? Can we set up a static ip for the lidar and all that?

JanezCim, 2022-02-03:
I did not check if I can setup the static IP, can you please try it and open an issue if it does not?

JanezCim, 2022-03-04:
With @rohbotics we agreed that the focal image creation is now at a level where we can release a kind of "release-candidate". This way we have some real world user feedback on issues that are going to be popping up. Not sure how to actually do that exactly, should i create dev and base images that have "release-candidate" in the name and maybe have the image print on every ssh a message, kindof: "This is not fully released image, some problems can still occour - please report problems onto : TODO which repo should they report problems to?"

JanezCim, 2022-03-11:
I will mark this as done, since we are releasing beta2 image soon

closed 2022-03-11 by JanezCim