UbiquityRobotics/ubiquity_main #262: Temporary LS lidar mount, getting it into production

MoffKalast, 2022-01-24:


JanezCim, 2022-01-31:
@AljazKrsnik currently the design of the small lidar holder is like this:
[image]

We need to rearrange the small holes to fit the new mounting holes of the LD lidar (green circles) @MoffKalast can you provide measures for those holes and a couple of pics of the lidar with the bottom plate? @AljazKrsnik I think those will become holes with M3 thread inserts, which is going to be easier to produce and easier to find the screws for. 

Stephen that wants to produce these wants this fairly soon, so if we could do it in the beginning of this week, it would be great, thanks!

AljazKrsnik, 2022-01-31:
Hello. Will do this by tomorrow.

MoffKalast, 2022-01-31:
Aight so, here are some pictures of the bottom plate for reference, and a reasonably accurate STL of it: https://workdrive.zohoexternal.com/folder/pykvle29eeb8d8c784bc584da07d6e40484bb

[image]

[image]

[image]

Note that arrow on the side in the last pic, that's the URDF forward direction for the lidar. I hope that helps.

AljazKrsnik, 2022-01-31:
Can you send me step file? With this file the scale is a bit f***** up. Just want to really get the correct measurement.
[image]

AljazKrsnik, 2022-01-31:
Is the orientation correct? I will send you more pics when this is confirmed.

MoffKalast, 2022-01-31:
There actually is no step file, I threw this together in tinkercad last week when I needed a printed mount for my test robot.

But that scale does look roughly right that you have there. This lidar is rather tiny. 

Holes should be M3, and here are the two main dimensions:

[image]

AljazKrsnik, 2022-01-31:
Ok. This is good enough. I just needed a scale. Thx

JanezCim, 2022-02-01:
A bit unrelated but still: @AljazKrsnik after this is done, would it be also possible to fit this into breadcrumb lidar holder?

AljazKrsnik, 2022-02-01:
No problem. Will try to make adjustments today

AljazKrsnik, 2022-02-01:
Please confirm the orientation of the lidar in both cases.
![ORIENT+CHARGING HOLE](https://user-images.githubusercontent.com/87651656/152018457-7793ec1b
[image]

[image]

[image]
-50ac-4614-b8e5-a9d670f2f336.GIF)

AljazKrsnik, 2022-02-01:
[image]

AljazKrsnik, 2022-02-01:
Regarding new/old matter- charging hole. Are there any changes( dimension vise)?

AljazKrsnik, 2022-02-01:
[image]

This is the right orientation. Am i right?

anfederman, 2022-02-01:
Is this for LS, LD06 or both?

AljazKrsnik, 2022-02-01:
Both

JanezCim, 2022-02-02:
@AljazKrsnik I think that yes, the forward arrow on lidar should be pointing into forward direction of the robot, so i think that this image is correct. but please @MoffKalast confirm this. 
[image]

This is not good, the lidar holder for small LD lidar will be mounted upside down. This way the small ld lidar will also be able to see between the bottom shelf and shell.
[image]

MoffKalast, 2022-02-02:
Yeah that looks good for the top pic, the red indicator should be facing the opposite way the LS lidar cable does.

AljazKrsnik, 2022-02-04:
The pictures show how much LIDAR mean is intersected in both aplications.
I think with the LIDAR N30105B v7 there is no problem.
With LIDAR 19, there is a lot of intersection.
My solution: I would lower the holes for LIDAR mount( shelf) on main cover for 9 mm. In that case there would be no beam intersection
[image]
.
Please correct me if i am wrong or my 3D model is different from phisical MAGNI 5

[image]
.

AljazKrsnik, 2022-02-07:
https://workdrive.zoho.com/home/g9vop387248f4f5304dde876eb982824d88c5/ws/g9vop0ead768c4254422e9332d4e4fd4a414b/folders/mskzr6baafd621a464ababc8453d77d2219ca

JanezCim, 2022-02-09:
This is not the correct termporary lidar holder, its breadcrumb lidar holder. What i need is the straight holder that goes into the lidar box

AljazKrsnik, 2022-02-09:
Right now, i have added thecorrect mount on the same link.

JanezCim, 2022-02-09:
the "small temporary lidar holder_TEST_HEX_NUT" you can probably remove from there since that holder is ment for breadcrumb and not for magni

JanezCim, 2022-02-10:
Status update: stephen from leishen has recieved the 3d file and we are in discussions about producing it

AljazKrsnik, 2022-02-11:
I have shaped a a mount JUST for LIDAR LD19. Its smaller and less rivets with a lot of clearance. Tell me what you think
[image]
[image]
[image]
.

JanezCim, 2022-02-11:
@AljazKrsnik this i think its safe to say its unrelated to the original issue of the temporary lidar mount, I opened up an issue here https://github.com/UbiquityRobotics/breadcrumb/issues/353 which is going to be specifically for the breadcrumb ls lidar mount

closed 2022-04-04 by MoffKalast