UbiquityRobotics/ubiquity_main #269: LS Lidar Metal Ethernet Plug

MoffKalast, 2022-03-07:
Can dangerously spark.

mjstn, 2022-03-09:
The ethernet cable connector for the Lidar has a large metal covering that is ground.    We plug this cable into our cpu in a very tight area where it is very easy to take out the cable or be putting it into the cpu and it can slip from a persons fingers and fall onto our main controller board where there are many different live voltages.     
[image]

If this cable were a typical plastic ended ethernet cable the risk of shorting out and causing damage would be greatly reduced.

NejaE, 2022-04-25:
Reported to Leishen, ordered a sample cable, should get an outdate by Apr 26th.

closed 2022-05-30 by MoffKalast