UbiquityRobotics/ubiquity_main #272: RootRework: Voronoi Graph Generator

MoffKalast, 2022-03-22:
We'll need a node that generates a mesh graph from an occupancy grid map using a voronoi graph approach. Each goal must be reachable by the robot and have some padding around it to allow for rotation.

The generated grid then needs to be integrated into the ezmap so it can then be adjusted for edge cases by a human operator.

I've assigned you here @JanezCim since you found some pre-existing code that we should take a look at and I can't recall what/where it was exactly, so if you still have those links it would be helpful to post them here as a reply.

JanezCim, 2022-03-23:
http://wiki.ros.org/tuw_voronoi_graph
http://wiki.ros.org/voronoi_planner
http://wiki.ros.org/dynamicvoronoi
https://github.com/lucascoelhof/voronoi_hsi

MoffKalast, 2022-04-12:
The tuw planner turned out to be the closest to what we need, currently forked as https://github.com/UbiquityRobotics/ubiquity_voronoi_graph to be reworked slightly.