UbiquityRobotics/ubiquity_main #287: Strage behaviour when controling GPIOs

ahrastnik, 2023-03-17:
Toggling or setting high as inputs, GPIOs 13 & 6 - which correspond to Cliff sensor inputs - result in RPi turning off WiFi, turning on the hot-spot or even powering off.

## Tested GPIO libraries
This behavior was observed with [pigpio](https://abyz.me.uk/rpi/pigpio/python.html#callback) & [wiringpi](http://wiringpi.com/) libraries. Therefore, the initial conclusion is it's not library dependent. However, while testing wiringpi, the pigpio was mistakenly running, so the test must be repeated without it; ideally, by making direct kernel calls.

## Affected images
So far, this behavior was observed on the following images:
- 2023-01-30-conveyorbot-focal-raspberry-pi
- ezmap image - TODO: Find out exact version

## TODO
1. [ ] Test *wiringpi* without the pigpio daemon running
2. [ ] Test all GPIOs on the RPi & log behaviors
3. [ ] Try toggling GPIOs by making direct kernel calls

ahrastnik, 2023-03-22:
So, I have done some research into this. So far, I haven't found anyone that has encountered this issue.
However, I have found a few leads inside Device Tree overlays:

- There is a "gpio-shutdown" config, which makes the Pi poweroff if the chosen GPIO changes state. But this doesn't explain why WiFi drops, as I found no config for toggling WiFi.
- The only place where GPIO 6 and 13 are used together is in hd44780-lcd display config. Otherwise, those GPIOs are used for two UART interfaces.

I have also found this issue on the [RPi forums](https://forums.raspberrypi.com/viewtopic.php?t=273032), which describes a similar issue, albeit on a completely different GPIO.

MoffKalast, 2023-03-22:
Hmm I've tried finding out which buttons the sonar board uses for its SW1 and SW2 but can't find the reference to it anywhere. Just to be sure, can you try commenting out "dtoverlay=ubiquity-led-buttons" from `/boot/config.txt` to see if that does anything?

ahrastnik, 2023-03-22:
Yikes, indeed we are, so much for the cliff sensor inputs... 😅

closed 2023-03-22 by ahrastnik