UbiquityRobotics/ubiquity_main #288: VNC unavailable

petunjiaj, 2023-03-22:
Hi, tank you for your support.
When I try to enable VNC by raspi-config, it gives me error.
There is a specific way to enable VNC?

versrion installed [2023-02-09-ubiquity-base-gdm3-focal](https://ubiquity-pi-image.sfo2.digitaloceanspaces.com/2023-02-09-ubiquity-base-gdm3-focal-raspberry-pi.img.xz)

anfederman, 2023-03-22:
Explain why you need to run a remote desktop? You can run rqt-image-view
remotely via either exporting the image topics or by ssh with the -XC
flags. A lot of overhead with VNC, and you don't want to load the robot up
with a GUI if you don't have to.

Furthermore, a GUI is available via http in both conveyorbot and EZmap.
Also, I am not sure that raspi-config. Which works with Raspian, works
fully with Ubuntu.

anfederman, 2023-03-22:
Also, I am not sure if the base image even has a desktop enabled. Hook the
Pi up to a monitor and see if a GUI is available.